prusa3d/Prusa-Firmware-Buddy #4449: Start Filtration when filament setting set to filter

ufgum, 2025-02-01:
### Printer model

XL

### Describe the enhancement

Why is there no automatic filtration option?
If the filament settings tells the filament needs filtration, the filtration starts automatically?

Post print filtration does!

`Features

    Filtration When you have enabled Enclosure and Print Filtration, the fan will operate at 80% of RPM. You can adjust the filtration performance in the menu (from 50% to 100%) during a running print job.

    Post Print Filtration You can also enable Post Print Filtration in Control -> Enclosure -> Manual Settings -> Post Print Filtration. When certain filaments (ASA, ABS, PC, FLEX, HIPS, PP) are detected in the G-code, the filtration will run for an additional 10 minutes to filter out all particles produced during printing. You can adjust the Post Print Filtration setting from 1 to 10 minutes..
    Note: PLA/PETG prints do not require post-print filtration.
`

### Expected functionality

the filtration starts automatically according to the filament settings

CZDanol, 2025-02-03:
Wdym? It is automatic. It says "when certain filaments are detected".

CZDanol, 2025-02-03:
My understanding is, the print filtration should start when you have "print filtration" enabled and a filament that requires filtration is used during the print.

ufgum, 2025-02-03:
Yes, but it behaves not like this.
- "print filtration" enabled: It is running on every print regardless the filament settings
- "post print filtration" enabled: It is running accordingly the filament settings 

So, bring both settings into line, or give the user 3 options: on/off/auto(filament settings)

CZDanol, 2025-02-03:
Okay, I've checked with my colleague and you are right, I am sorry.

We are currently implementing a filtration system for the CORE One as well and we will be joining the two functionalities, so I will raise this up.

The main point for the automatic filtration working only post print is that you don't really need to filter air that much during the print, because you're not opening the enclosure and filtering would reduce the chamber temperature, because the fans blow the air out. So, you basically want to only clear the air after the print has finished, before the user opens the enclosure - and that's why the automatic part is only for the post print.

But it would still make sense to run the fans on some minimal level even during the print only to prevent the fumes seeping out through the cracks.

ufgum, 2025-02-03:
> But it would still make sense to run the fans on some minimal level even during the print only to prevent the fumes seeping out through the cracks.

Yes but herefore we need one more option:
- fan speed while printing
- fan speed post printing

I like the on/off/auto setting for printing and post printing, so anybody can adapt to his needs.

Gik-0101, 2025-02-03:
I absolutely agree. I also knew this from my Voron. 
During printing approx. 20% fan speed as this prevents the particles from escaping from the enclosure. Not vacuuming only works if the enclosure is absolutely tight. The XL definitely doesn't have that.
A post-pressure filter can then run at a higher fan speed. 
The enclosure fan should also not run during the warm-up phase and start when the printer starts to draw the intro line.

CZDanol, 2025-02-04:
Internal ticket: BFW-6719

Prusa-Support, 2025-11-22:
Full integration and extended settings for the Filtration have been implemented on Prusa CO starting from FW 6.3.1, and ported to the other printer models toghether with cumulative improvements with FW 6.4.0-alpha (not recommended for Prusa XL).

Please try the most recent pre-release FW 6.4.0-RC2.
https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.4.0-RC2

Michele Moramarco
Prusa Research

ex-nerd, 2025-11-30:
I just updated to the 6.4.0 final release on my XL with ASA and seemingly can't get the filtration fan to run at all. I think this is because of the "heated chamber" setting but the XL enclosure isn't airtight so there are fumes leaking out (I can smell them, they're causing a headache, and my VOC sensor has been steadily climbing into the "not healthy" range)… should at least be a way to set the fan to something like 10-20% to make sure any air leaking out of the enclosure goes through the charcoal filter.

ufgum, 2025-11-30:
> > Did you try a setting higher than 25%?
> 
> Yeah. I'd been running it at 100% for some smelly PLA (love those random color additives), so it was set that high when I started. As soon as I switched to ASA with its heated-enclosure settings, I couldn't get the chamber fan to turn on at all during the print no matter what value I changed it to (during the print).
> 
> Unfortunately, there doesn't seem to be a way to override the chamber temperature settings mid-print, or I would have checked that to confirm.

In mid print I could only submit my changes if I choose disable encloser fan at all and reenable it. Some strange behavior I need to check again to give more detailed report here.

ex-nerd, 2025-12-12:
> [@ex-nerd](https://github.com/ex-nerd) Unfortunately, Prusa XL doesn't use a chamber sensor ([#3848](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3848)) but an estimated chamber temperature for safety features. In fact, our stock profiles in Prusa Slicer only use chamber nominal/minimal parameters where it makes sense, in case of Prusa CORE One - [#4722 (comment)](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4722#issuecomment-3533174229).

Thanks. #3848 explains some of the confusion here. This really is something that should be included in PrusaSlicer and in the online documentation (both for PrusaSlicer and the enclosure) and on the product page for the enclosure, which currently states `You can check the temperature inside the Enclosure … using the information on the printer display` (which really implies that this is an accurate value and risks running afoul of truth-in-advertising regulations)

> Chamber temperature parameters are not applied on profiles for Prusa XL. Prusa XL Enclosure is not designed for temperature monitoring, and not proven to operate at high temperatures. These chamber temperature parameters are not meant to control not the filtration fan itself but the behavior of chamber cooling fans - unless filtration fan replaces chamber fans, which is the case of Prusa CO, again.

I'm not seeing this. Or rather, I _do_ see that the gcode when I print ASA includes both `M191 S40` and `M141 S60` at the start of the print, and the printer delays with a message like "waiting for chamber to reach temperature" before it starts to print.

Other thoughts:

- I rarely use extruder 1 for exotic filaments. Is it possible that this is an issue of the printer reading `; filament_type = PLA;PLA;PLA;ASA;PLA` in the gcode but only triggering the filter based on the first extruder instead of the one actually used during the print? (if so, this has its own problems due to #5017)
- I would expect the chamber filtration settings to be set based on the `Filament type` property, not whatever random name I give the filament. The only thing I've changed is the custom gcode to slow down cooling to prevent warping: [image]
  - Regardless, I was using the stock `Buddy3D ASA` and `Prusament ASA` profiles before I duplicated the Prusament one to add the custom gcode. The fan settings behave as previously described.
- The chamber fan seems to start correctly (well, "correctly") as long as I have "filter all materials" enabled. Of course, the whole point of this feature is  that I shouldn't need that, so I now have to toggle the setting on/off manually when I print PLA vs ASA.
- If it's a known issue that the fan needs an extra kick of power to start spinning, this seems like something that could be adjusted in the firmware itself. E.g. if the requested fan speed N% is below 30%, spin to 50% for 1s and then reduce power to N%.

ex-nerd, 2025-12-14:
I just tried a print using the built-in `Buddy3D ASA` profile on E1 and E4, with the print actually using only E4. "Filter all materials" is disabled, and the chamber fan started up as expected (though it waited until all of the "warmup" like bed leveling, nozzle cleaning, etc).

Unfortunately for the "first tool" theory, the fan also worked as expected when printing with `Generic PLA` in tool 1. And … also when using my Polymaker ASA profile on tool 4. It's acting like this issue has fixed itself, so 🤷‍♂️🫠. The only change I made was set the chamber fan speed to 60% instead of the 30% I normally use for ASA, but I had it at 100% before I started all of this so I don't see how that's related.

ex-nerd, 2026-09-28:
> This issue has not received any updates for 60 days

Still waiting for Prusa to reply…