prusa3d/Prusa-Firmware-Buddy #4450: [BUG] No module named 'requests'

espr14, 2025-02-01:
### Printer model

all

### Firmware version

6.2.0

### Printing from...

building

### Describe the bug

New version 6.2.0 uses Python package `requests` in `utils\bootstrap.py` which is not installed and installing does not help.

### How to reproduce

```
python utils/build.py --bootloader empty --preset mini-en-cs --final
Building mini-en-cs_release_emptyboot
-- Python3: C:/git/P32-FW/.venv/Scripts/python.exe
-- The CXX compiler identification is GNU 13.2.1
-- Check for working CXX compiler: C:/git/P32-FW/.dependencies/gcc-arm-none-eabi-13.2.1/bin/arm-none-eabi-g++.exe
Traceback (most recent call last):
  File "C:\git\P32-FW\utils\bootstrap.py", line 25, in <module>
    import requests
ModuleNotFoundError: No module named 'requests'
CMake Error at C:/git/P32-FW/cmake/Utilities.cmake:32 (message):
  Failed to obtain recommended gcc version from utils/bootstrap.py
Call Stack (most recent call first):
  C:/git/P32-FW/cmake/GccArmNoneEabi.cmake:6 (get_recommended_gcc_version)
  C:/git/P32-FW/build/mini-en-cs_release_emptyboot/CMakeFiles/3.28.3/CMakeSystem.cmake:6 (include)
  C:/git/P32-FW/build/mini-en-cs_release_emptyboot/CMakeFiles/CMakeScratch/TryCompile-3fydgs/CMakeLists.txt:2 (project)

CMake Error: CMAKE_CXX_COMPILER not set, after EnableLanguage
CMake Error at .dependencies/cmake-3.28.3/share/cmake-3.28/Modules/CMakeTestCXXCompiler.cmake:49 (try_compile):
  Failed to configure test project build system.
Call Stack (most recent call first):
  CMakeLists.txt:9 (project)

-- Configuring incomplete, errors occurred!

Building finished: 0 success, 1 failure(s).
```

### Expected behavior

```
Building finished: 1 success, 0 failure(s).
```

rcloran, 2025-02-02:
https://gist.github.com/rcloran/56f89281c9aa321257cf6164dfc505bd works around this for me ... I'm not sure if it's a correct fix, so I'm hesitant to send a PR with it, though.

Tomcus, 2025-02-03:
Hi thanks, for reporting the issue. We know of these issues and fixed them in our private repository and they will be fixed in `6.2.1` or in `6.3.1` depending on what is released first :). @rcloran's solution is bit hacky, but just installing the `requests` package into the virtual environment.

The easiest solution for you now should be running:
```bash
.venv/bin/pip3 install requests
```

Tomcus, 2025-03-28:
@seppelicous Do you have `requests` installed locally? You must install it before running `build.py` now.

seppelicous, 2025-03-28:
Actually they are installed. I just cloned a fresh copy of the git.
Please see the output below:

`➜  Prusa-Firmware-Buddy git:(master) .venv/bin/pip3 install requests    
Requirement already satisfied: requests in ./.venv/lib/python3.13/site-packages (2.32.3)
Requirement already satisfied: charset-normalizer<4,>=2 in ./.venv/lib/python3.13/site-packages (from requests) (3.4.1)
Requirement already satisfied: idna<4,>=2.5 in ./.venv/lib/python3.13/site-packages (from requests) (3.10)
Requirement already satisfied: urllib3<3,>=1.21.1 in ./.venv/lib/python3.13/site-packages (from requests) (2.3.0)
Requirement already satisfied: certifi>=2017.4.17 in ./.venv/lib/python3.13/site-packages (from requests) (2025.1.31)

[notice] A new release of pip is available: 23.3.2 -> 25.0.1
[notice] To update, run: /Users/Sebastian/Prusa-Firmware-Buddy/.venv/bin/python -m pip install --upgrade pip
➜  Prusa-Firmware-Buddy git:(master) python3 utils/build.py --preset mk4
Switching to Buddy's virtual environment.
You can disable this by setting the BUDDY_NO_VIRTUALENV=1 env. variable.
Building mk4_release_boot
-- Python3: /Users/Sebastian/Prusa-Firmware-Buddy/.venv/bin/python3.13
-- The C compiler identification is GNU 13.2.1
-- The CXX compiler identification is GNU 13.2.1
-- The ASM compiler identification is GNU
-- Found assembler: /Users/Sebastian/Prusa-Firmware-Buddy/.dependencies/gcc-arm-none-eabi-13.2.1/bin/arm-none-eabi-gcc
-- Check for working C compiler: /Users/Sebastian/Prusa-Firmware-Buddy/.dependencies/gcc-arm-none-eabi-13.2.1/bin/arm-none-eabi-gcc
Traceback (most recent call last):
  File "/Users/Sebastian/Prusa-Firmware-Buddy/utils/bootstrap.py", line 25, in <module>
    import requests
ModuleNotFoundError: No module named 'requests'
CMake Error at /Users/Sebastian/Prusa-Firmware-Buddy/cmake/Utilities.cmake:32 (message):
  Failed to obtain recommended gcc version from utils/bootstrap.py
Call Stack (most recent call first):
  /Users/Sebastian/Prusa-Firmware-Buddy/cmake/GccArmNoneEabi.cmake:6 (get_recommended_gcc_version)
  /Users/Sebastian/Prusa-Firmware-Buddy/build/mk4_release_boot/CMakeFiles/3.28.3/CMakeSystem.cmake:6 (include)
  /Users/Sebastian/Prusa-Firmware-Buddy/build/mk4_release_boot/CMakeFiles/CMakeScratch/TryCompile-Pn7Lh8/CMakeLists.txt:2 (project)

CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
CMake Error at .dependencies/cmake-3.28.3/share/cmake-3.28/Modules/CMakeTestCCompiler.cmake:56 (try_compile):
  Failed to configure test project build system.
Call Stack (most recent call first):
  CMakeLists.txt:9 (project)

-- Configuring incomplete, errors occurred!
Building mk4_release_noboot
-- Python3: /Users/Sebastian/Prusa-Firmware-Buddy/.venv/bin/python3.13
-- The C compiler identification is GNU 13.2.1
-- The CXX compiler identification is GNU 13.2.1
-- The ASM compiler identification is GNU
-- Found assembler: /Users/Sebastian/Prusa-Firmware-Buddy/.dependencies/gcc-arm-none-eabi-13.2.1/bin/arm-none-eabi-gcc
-- Check for working C compiler: /Users/Sebastian/Prusa-Firmware-Buddy/.dependencies/gcc-arm-none-eabi-13.2.1/bin/arm-none-eabi-gcc
Traceback (most recent call last):
  File "/Users/Sebastian/Prusa-Firmware-Buddy/utils/bootstrap.py", line 25, in <module>
    import requests
ModuleNotFoundError: No module named 'requests'
CMake Error at /Users/Sebastian/Prusa-Firmware-Buddy/cmake/Utilities.cmake:32 (message):
  Failed to obtain recommended gcc version from utils/bootstrap.py
Call Stack (most recent call first):
  /Users/Sebastian/Prusa-Firmware-Buddy/cmake/GccArmNoneEabi.cmake:6 (get_recommended_gcc_version)
  /Users/Sebastian/Prusa-Firmware-Buddy/build/mk4_release_noboot/CMakeFiles/3.28.3/CMakeSystem.cmake:6 (include)
  /Users/Sebastian/Prusa-Firmware-Buddy/build/mk4_release_noboot/CMakeFiles/CMakeScratch/TryCompile-Lmq9yL/CMakeLists.txt:2 (project)

CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
CMake Error at .dependencies/cmake-3.28.3/share/cmake-3.28/Modules/CMakeTestCCompiler.cmake:56 (try_compile):
  Failed to configure test project build system.
Call Stack (most recent call first):
  CMakeLists.txt:9 (project)

-- Configuring incomplete, errors occurred!

Building finished: 0 success, 2 failure(s).
 mk4_release_boot   project configuration FAILED
 mk4_release_noboot project configuration FAILED`

seppelicous, 2025-03-28:
Now I downloaded the source code of the 6.2.3 release and here I it works better after installing requests in the vent.
But I run into another issue and I don't know if it is actually related to the requests package

`prusa623 python3 utils/build.py --preset mk4
Switching to Buddy's virtual environment.
You can disable this by setting the BUDDY_NO_VIRTUALENV=1 env. variable.
Downloading gcc-arm-none-eabi-13.2.1 done
Extracting arm-gnu-toolchain-13.2.rel1-darwin-x86_64-arm-none-eabi.tar.xzg Traceback (most recent call last):
  File "/Users/Sebastian/prusa623/utils/build.py", line 828, in <module>
    main()
    ~~~~^^
  File "/Users/Sebastian/prusa623/utils/build.py", line 789, in main
    bootstrap.bootstrap()
    ~~~~~~~~~~~~~~~~~~~^^
  File "/Users/Sebastian/prusa623/utils/bootstrap.py", line 329, in bootstrap
    install_dependency(dependency)
    ~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^
  File "/Users/Sebastian/prusa623/utils/bootstrap.py", line 236, in install_dependency
    download_and_unzip(url=url, directory=installation_directory)
    ~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/Sebastian/prusa623/utils/bootstrap.py", line 177, in download_and_unzip
    with zipfile.ZipFile(filename, 'r') as obj:
         ~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/Library/Frameworks/Python.framework/Versions/3.13/lib/python3.13/zipfile/__init__.py", line 1385, in __init__
    self._RealGetContents()
    ~~~~~~~~~~~~~~~~~~~~~^^
  File "/Library/Frameworks/Python.framework/Versions/3.13/lib/python3.13/zipfile/__init__.py", line 1452, in _RealGetContents
    raise BadZipFile("File is not a zip file")
zipfile.BadZipFile: File is not a zip file`

Tomcus, 2025-03-28:
From the output you show me that the package is installed in the virtual environment not system wide (you now need that it is mentioned in README). Try running the build.py inside the venv, that should do it for you. Sorry for the troubles nobody uses MAC in our team and it works fine for Linux or Windows developers (although they both use WSL). If running build.py in venv doesn't help, edit the build.py, catch the ModuleNotFound error and print which python was used to call the script/what package folder did it use.

seppelicous, 2025-03-28:
Unfortunately, my python knowledge is very limited and therefore I don't know how to run it inside a venv.
Since it works with the 6.2.3 source, at least I don't get the "requests" error, is the second error I mentioned above related to any request package in your opinion?

seppelicous, 2025-03-29:
So, I have set up a Debian 12 VM and tried it again.
Even when venv is activated and requests is actually installed in the virtual environment, it thorws this error:

```
(venv) debian@debian:~/prusa623$ pip3 install requests
Requirement already satisfied: requests in ./venv/lib/python3.11/site-packages (2.32.3)
Requirement already satisfied: charset-normalizer<4,>=2 in ./venv/lib/python3.11/site-packages (from requests) (3.4.1)
Requirement already satisfied: idna<4,>=2.5 in ./venv/lib/python3.11/site-packages (from requests) (3.10)
Requirement already satisfied: urllib3<3,>=1.21.1 in ./venv/lib/python3.11/site-packages (from requests) (2.3.0)
Requirement already satisfied: certifi>=2017.4.17 in ./venv/lib/python3.11/site-packages (from requests) (2025.1.31)
(venv) debian@debian:~/prusa623$ sudo python3 utils/build.py --preset mk4
Switching to Buddy's virtual environment.
You can disable this by setting the BUDDY_NO_VIRTUALENV=1 env. variable.
Python executable (/home/debian/prusa623/.venv/bin/python) is missing the "requests" package.
Traceback (most recent call last):
  File "/home/debian/prusa623/utils/build.py", line 27, in <module>
    import bootstrap
  File "/home/debian/prusa623/utils/bootstrap.py", line 26, in <module>
    import requests
ModuleNotFoundError: No module named 'requests'
(venv) debian@debian:~/prusa623$ which python3
/home/debian/prusa623/venv/bin/python3
(venv) debian@debian:~/prusa623$ python3 --version
Python 3.11.2
(venv) debian@debian:~/prusa623$ 

```

rcloran, 2025-03-29:
> ```
> (venv) debian@debian:~/prusa623$ pip3 install requests
> Requirement already satisfied: requests in ./venv/lib/python3.11/site-packages (2.32.3)
> ...
> Python executable (/home/debian/prusa623/.venv/bin/python) is missing the "requests" package.
> ...
> (venv) debian@debian:~/prusa623$ which python3
> /home/debian/prusa623/venv/bin/python3
> ```

In this case, you've created the venv at `./venv`, but the build system is using a venv at `./.venv`. When you run the build through sudo, it's likely (I say likely because sudo can be configured in different ways) not passing the relevant environment variables for the venv. 

I suggest you don't build as root, there's no need. I also suggest that you install requests in the same venv as the Prusa build system is expecting to use. 

For the invalid zip file problem on macOS, see #4451

espr14, 2025-04-19:
In 6.3.2 works just fine.

espr14, 2025-04-21:
Still in v6.3.2 :-(

CZDanol, 2025-04-22:
Thee's the `requirements.txt` file for pip that is afaik a standard way to say which packages are required. Calling pip from the python itself doesn't seem like a clean solution to me.

espr14, 2025-04-23:
However, clean repository of v6.3.2 works as is. So the issue can be closed.

CZDanol, 2025-04-24:
Aight.

closed 2025-04-24 by CZDanol