prusa3d/Prusa-Firmware-Buddy #4454: [ENHANCEMENT]

bryn51, 2025-02-03:
### Printer model

MK4, MK4S

### Describe the enhancement

Current Behaviour: The user sends a file to the printer.  In some conditions the printer may cease printing at some point in the file.  OR the printer may display a BSOD.
The remedies available to the user:
a) delete one or more files on the USB
b) reinitialise the USB filesystem

And what many do is to simply replace the USB, not knowing how to deal with the issue.  This leads to added operational cost and downtime for affected users and printers.

Note: "A FAT32 directory can have 65,536 directory entries. **Each file and subdirectory takes from 2 to 13 entries**, depending on the length of its name. It is not possible to change the maximum number of entries a directory can store.".  That limit is what applies on a window platform.  But the OS of the Buddy board is different, and its limits are unknown.

The standard USB has a capacity of 32 GBYtes, but the error occurs well before the 32 GBytes capacity is reached. 
The error also occurs well before the limit on directory entries is reached. But it seems possible that the implementation of FAT32 on the Buddy Board has some lesser capacity.

Prusa connect also will happily attempt to send a file to the printer that has an identical name to a file that exists on the printer. 

To date, PC and printer firmware support subdirectories when made manually by the user, but do not automatically place files into folders.
Mostly, uploaded files are stored in the base folder of the USB, and it appears like there is no management of the number of files, and total size of subdirectory storage capacity. 

Furthermore, Prusa currently provide little or no information about any such limitations or suggest any strategy for USB file management by the user.

### Expected functionality

Whatever the case, this enhancement proposes that the condition be investigated, and some better management of USB resource  is made by printer firmware or PC (or both) and Prusa Slicer needs to be also handled.
  This is an obvious misoperation of PC that should be addressed. Its pretty simple to devise a system of making unique file names.
I note that Prusa Slicer cannot know what files exist on the printer. It tries to make a unique file name but often enough successive slice operations result in non unique names. Again, its pretty simple to devise a system of guaranteed unique file names that use a UUID or similar technique.  The best technique is the simplest technique.

This proposal suggests that active management of limitations of USB FAT32 directory capacity be made by printer firmware with or without involvement of PrusaConnect.  So that the limitations of FAT32 USB filesystem are not encountered in normal use by the user.
- automatic adjustment of filenames to make them unique when sent to the printer.
- alert the user of prusa connect when a printers USB filesystem needs maintenance attention by the user: 
-  a prompt or message telling them to reduce the number of files on the USB, or deploy folders.
- the sending of a file from prusa slicer to prusa connect to the printer needs to continue to be seamless, with error conditions automatically handled.

 The  information gap needs to be addressed by way of a kbase help article.  A usb filesystem management strategy needs to be suggested for the benefit of users that are not necessarily fully understanding of IT terminology and filesystem managament.

bryn51, 2025-04-07:
Please do not sideline this issue.  Its really important to have better control over the files stored on the Printer USB. What I'd like you to do is escalate it, not cancel it.

CZDanol, 2025-04-07:
Hello,
first of all, please add a title to your issue.

Second, after reading it, I don't really understand what this is about?
Is this about filename conflict when uploading from Connect? That has already been resolved on the Connect side, they are adding [x] suffix.

Is this about exhausting FAT32 entries? Has this ever occured in practice?

Or what am I missing?

bkerler, 2025-04-13:
If I understand correctly, the issue happens due to having the same filename and is caused not by the printer firmware, but by PrusaSlicer. It can be easily fixed by changing the output filename and adding "_{timestamp}" to it. It has nothing to do with FAT FS.

bkerler, 2025-04-13:
I wrote a PR [at Prusaslicer Repo](https://github.com/prusa3d/PrusaSlicer/pull/14438) for this issue.

closed 2025-06-23 by github-actions[bot]