prusa3d/Prusa-Firmware-Buddy #4458: [BUG]

bigman212-help, 2025-02-05:
### Printer model

MK3.5 and MMU3

### Firmware version

6.2.0 and MMU 3.0.3

### Printing from...

Prusa Connect

### Describe the bug

Prusa connect displays 5 nozzle temperatures when printing with MMU

### How to reproduce

Enable MMU and look at prusa connect

### Expected behavior

Single nozzle temperature.

danopernis, 2025-02-05:
@bigman212-help thanks for the report. Is this a new issue, or was it already present in 6.1.x version?

CZDanol, 2025-02-05:
I've asked the Connect guys about this

bigman212-help, 2025-02-05:
Just upgraded the printer today. Factory reseted the printer to be sure and this keeps happening.

Only fix was to disable mmu. Although when I turned off the MMU and then turned it on it gave a MMU overcurrent warning on the power on state intermittently.

CZDanol, 2025-02-06:
Connect confirmed that this is an issue on their side.

closed 2025-04-15 by github-actions[bot]