prusa3d/Prusa-Firmware-Buddy #4459: [ENHANCEMENT] Expose chamber states in PrusaLink UI & API

muesli, 2025-02-05:
### Printer model

Core One

### Describe the enhancement

The current PrusaLink UI & API exposes all kinds of states and sensor readings, e.g. for the bed, tool, material etc.

With the Core One being out, I noticed that the current chamber settings & state are not part of the web UI or the API response. This would be a great addition.

### Expected functionality

Expose chamber temperature setting & sensor readings in PrusaLink UI & API.

PinkPandaKatie, 2025-02-20:
I have some code to add the temperature to the API but I haven't added it to the UI.
In `/lib/WUI/nhttp/status_renderer.cpp`:

Add to top:
```c++
#if HAS_CHAMBER_API()
#include <feature/chamber/chamber.hpp>
#endif
``` 

Then add just after "target_nozzle":
```c++
#if HAS_CHAMBER_API()
            JSON_FIELD_FFIXED("temp_chamber", buddy::chamber().current_temperature().value_or(0), 1) JSON_COMMA;
            JSON_FIELD_FFIXED("target_chamber", (buddy::chamber().capabilities().temperature_control() ? buddy::chamber().target_temperature() : std::nullopt).value_or(0), 1); JSON_COMMA;
#endif
``` 

I pushed a branch with just this change but I haven't done a PR yet: https://github.com/PinkPandaKatie/Prusa-Firmware-Buddy/tree/prusalink-api-chamber-temperature

muesli, 2025-04-22:
Stiil relevant.

mkyral, 2025-05-08:
Just build Core One kit and I really like to monitor all temperatures in Homeassistant. I'm surprised, that this is not available yet.

mkyral, 2025-09-15:
Dear BOT. This issue is still valid.

azerioxal, 2026-01-04:
The chamber temperatures and fan speeds would be great inclusions to PrusaLink.

xNymia, 2026-08-27:
Would love to see this, Enclosure and Heatbreak temps are important if printing PLA in a warm location. Being able to monitor it remotely would be great.

trimeta, 2026-08-27:
Let's be honest: with the announced plans to replace PrusaLink with [Prusa Connect Local](https://www.prusa3d.com/p/prusa-connect-local/early-access/), they're never going to add new features to PrusaLink.

mkyral, 2026-08-27:
But API is still there. Does not matter how is frontend application named,

mkyral, 2026-10-27:
Still valid.

endrift, 2026-10-27:
Stale bot continues to be annoying as hell

wirgakatz, 2026-11-01:
I would also be super interested in seeing this. Especially having the chamber temperature and chamber temperature target exposed in the api would be super useful