prusa3d/Prusa-Firmware-Buddy #4461: [BUG] No phase stepping in the Settings menu(CoreOne)

CloudStrife1983, 2025-02-05:
### Printer model

Core One

### Firmware version

6.3.0

### Printing from...

N.A.

### Describe the bug

In the Settings Menu, there is no Phase Stepping but it should be. 

[image]

[image]

### How to reproduce

Navigate in the Settings Menü. Scroll through the Menu.

CZDanol, 2025-02-06:
Hello, phase stepping is always enabled on the Core ONE and it is not possible to disable it (because it wouldn't server any purpose). I don't know what the screenshot is from, but it's the document that's out of spec, not the firmware.

closed 2025-02-06 by CZDanol

blgli, 2025-02-06:
So I got my core one yesterday, and I believe you are mistaken that phase stepping is turned on by default with no option to disable it. My core one printing the included lighthouse file is VERY loud, so loud that I heard it from the lower story of my house. I had to stop the print because there was no way I was going to listen to it for 22 hours. It is a lot louder than my MK4s and even my vorons. I am on firmware 6.3.0+10073. 

I am wonder if different firmware was sent to people with review units because a few of those users claim that the printer is “so quiet” but I have completely opposite experiences.

cristian-rivera, 2025-02-06:
> So I got my core one yesterday, and I believe you are mistaken that phase stepping is turned on by default with no option to disable it. My core one printing the included lighthouse file is VERY [...]

I agree. It is way louder than my MK4S. Input Shaper calibration helped a little bit, but I'm hoping there are some future updates to address the sound.

blgli, 2025-02-06:
Looks like the X/Y motors are 0.9 degree motors, and according to the person above they do not have phase stepping setup for them yet. So then the core one doesn’t have phase stepping?

Cause the XL uses 1.8 degree motors

CloudStrife1983, 2025-02-06:
> So I got my core one yesterday, and I believe you are mistaken that phase stepping is turned on by default with no option to disable it. My core one printing the included lighthouse file is VERY [...]

 I can't confirm that. Yes, the frequency during fast movements is slightly louder/higher, but this is due to the rail on X. It's the same with the XL. At normal speeds, my Core One is actually pretty quiet.

CZDanol, 2025-02-08:
>  I believe you are mistaken that phase stepping is turned on by default with no option to disable it.

It is turned on on the Core one. But when uncallibrated, it does not have any effect nothing in terms of loudness – but it's computationally easier to drive the motors this way. When you will be able to calibrate it, it should silence the printer quite nicely.

CloudStrife1983, 2025-02-08:
> It is turned on on the Core one. But when uncallibrated, it does not have any effect nothing in terms of loudness – but it's computationally easier to drive the motors this way. When you will be [...]

I'm just asking because my Core One vibrates quite a bit when I apply certain pressure. It sounds just like the XL at the beginning when the Stepping Calibration phase did not yet exist. 

You can listen to/watch it here under this link: 

https://drive.google.com/file/d/1-NYn1wOrUK8HDYgAOzgrPsmcLKOG4P69

blgli, 2025-02-08:
> It is turned on on the Core one. But when uncallibrated, it does not have any effect nothing in terms of loudness – but it's computationally easier to drive the motors this way. When you will be [...]

Well that’s kind of useless and misleading. I know the product description of the Core One mentions to have phase stepping but clearly it doesn’t. What is the timeline for this to be fully implemented into firmware?

TheDuchy, 2025-03-18:
I don't wanna be THAT guy, but is there any approximate ETA on the calibration feature getting shipped?

CZDanol, 2025-03-18:
I am sorry, but I cannot publish any estimates at this moment. But we are genuinely working on it.

rinkek, 2025-03-30:
Same issue on my Core One, it resonates in the metal construction somewhere. This makes alot of noise.
I replaced quite a few of the rivits for screws and nylocks. No luck so far.

blgli, 2025-03-30:
> Same issue on my Core One, it resonates in the metal construction somewhere. This makes alot of noise. I replaced quite a few of the rivits for screws and nylocks. No luck so far.
> 
> I increased the speed with 110%, this helped a bit with resonances. Now i speed with structural profile for inner walls is 88mm/s

You’re not alone. Ive seen so many people saying that the printer is “quiet” in reviews. In normal mode, this is the noisiest machine I own by a long shot. It’s very loud and can be heard across my house.

rinkek, 2025-03-31:
Just hope there will be an update here soon...

rinkek, 2025-04-02:
I just installed 6.3.1 and it's not in this update yet, hopefully next release.

danopernis, 2025-04-02:
Yes, that is correct, we had to disable the calibration in the menu for the final build. It fails too often and we didn't want to deal with all the bug reports due to BSOD caused by OOM situation.

If you are running 6.3.1 and you still want to try it, you can issue M1977 G-Code via serial line or by "printing" a file containing just this line. This should run the wizard. That being said, please understand that this is still experimental and may fail in unexpected ways. If it fails, you should be able to recover by doing full factory reset.

We are in a weird spot here, where we would like to gather more feedback from the community, yet this feature may actually harm the printer's performance. I sincerely hope that github users would understand this delicate situation.

danopernis, 2025-04-02:
To reset only the phase stepping calibration, it should be enough to issue `M1977 D`

Still, this is experimental, untested, and you might need to do full factory reset if anything goes wrong. Also, these G-Codes might change in the future. If anybody still wants to try that, I will be happy for the feedback.

rinkek, 2025-04-02:
But can i still access the menu when something goes wrong?
I don't mind doing a factory reset

danopernis, 2025-04-02:
All of the G-codes I mentioned only affect the phase-stepping corrections to motor currents, so all the basic functionalities of the firmware (such as accessing the menus) should remain intact. You should always be able to revert to factory settings.

blgli, 2025-04-02:
> Yes, that is correct, we had to disable the calibration in the menu for the final build. It fails too often and we didn't want to deal with all the bug reports due to BSOD caused by OOM situation. [...]

Just gave it a try on my CORE One, no luck at all. Machine homes, asks me to put the accelerator on the nozzle, I hit continue, it goes to start the X calibration, does one diagonal move and the machine reboots and I get the memory heap error 

Shame I can’t even attempt to calibrate.

rinkek, 2025-04-02:
Same here, attached the accelerometer, motor moves, then reset and blue screen with "Out of heap memory"

danopernis, 2025-04-02:
@blgli yeah, that's why we wanted to wait until we enable it for general public. Too unreliable at this point. Memory is particularly fragmented after the update, so reset might help. We are currently testing some memory optimizations as well, and it looks promising.

rinkek, 2025-04-02:
> [@blgli](https://github.com/blgli) yeah, that's why we wanted to wait until we enable it for general public. Too unreliable at this point. Memory is particularly fragmented after the update, so reset might help. We are currently testing some memory optimizations as well, and it looks promising.

I have used the reset button a few times, but did not help. Would a factory reset do anything to make it work?

blgli, 2025-04-02:
> I have used the reset button a few times, but did not help. Would a factory reset do anything to make it work?

I think a factory reset is what he means. Gonna give it a shot, I need this printer to be quiet so I can actually use it

danopernis, 2025-04-02:
No. If running the calibration after reboot (no factory reset needed) doesn't work, please wait for the next firmware version. We are on the edge of the hardware capabilities, we have some optimuzations ready, please stay tuned.

rinkek, 2025-04-02:
Tried it around 6 times, no luck.

blgli, 2025-04-02:
> Tried it around 6 times, no luck.

Yup same here, even went to the extreme of resetting the printer to factory defaults and installing firmware again from the usb drive. No luck. Guess I’ll just have to wait for the next release….

rinkek, 2025-04-02:
isn't there any nightly build we can try?

rinkek, 2025-04-03:
> No. If running the calibration after reboot (no factory reset needed) doesn't work, please wait for the next firmware version. We are on the edge of the hardware capabilities, we have some optimuzations ready, please stay tuned.

Not tuned yet :)

rinkek, 2025-04-05:
Any news?
Can we try something?

e-basteln, 2025-04-14:
> we have some optimizations ready, please stay tuned.

Yet another firmware release (6.3.2), still no phase stepping calibration.
Is there still hope to get this implemented robustly?

rinkek, 2025-04-14:
> Still working on this. Out of memory issues have been solved. The algorithm is also more robust. Steady progress, just takes more time than anticipated.
> 
> To prevent confusion, 6.3.2 only contains the fan speed threshold adjustments, none of the changes mentioned above.

^- keeps F5-ing

CZDanol, 2025-04-15:
Nope.  C1 will come out first, then we will possibly start looking into making it run on MK4.

Zerschmelzer3000, 2025-05-01:
I'm a bit dissapointed to read that. Please enable phase stepping on the MK4/S too!

Takrel, 2025-05-03:
Would also like to have phase stepping available. The printer is pretty loud on some movements.

andersnygaard, 2025-05-19:
Would love to see phase stepping if this is the solution to the VFA problem. See attached [image of a part printed with PETG on a Core One](https://imgur.com/a/VoADAZf)

slackista, 2025-05-20:
>Would love to see phase stepping if this is the solution to the VFA problem. See attached [image of a part printed with PETG on a Core One](https://imgur.com/a/VoADAZf)

I've got similar results with MK4S. Printing with `Structural` profile causing printer annoying noisy resonance.

rinkek, 2025-05-20:
Would be nice if we could get an update on this.

fyby, 2025-05-21:
Would love to have PS on C1. Just upgraded from mk4s and its a helluva sound monster now :-D

Takrel, 2025-05-22:
I would also like to help gathering more data and testing an alpha- or beta-FW for my Core One.

fyby, 2025-05-22:
I can too do some tests if you want. I will gladly help.

TrueScott, 2025-05-23:
Would it be possible to release an alpha firmware for those who want to test and help?

rinkek, 2025-05-23:
The 

> > Would it be possible to release an alpha firmware for those who want to test and help?
> 
> I believe you can still run gcode `M1977 D` to test it out, though from my initial code review of 6.3.3 I think the phase stepping / accelerometer gcode (among other things) has been reworked. Perhaps one of the devs could check back in? People want to help!
> 
> [@danopernis](https://github.com/danopernis) 🙏

I don't think 6.3.3 contains anything new related to phase stepping.
I tested M1977 on 6.3.2 and got the memory error every time, so i'm pretty sure i'll get this error on 6.3.3 too.

atuleu, 2025-05-23:
> I don't think 6.3.3 contains anything new related to phase stepping. I tested M1977 on 6.3.2 and got the memory error every time, so i'm pretty sure i'll get this error on 6.3.3 too.

It does. See 20bf392c8fabacb0b1abcb39534ea4a633fa099e who enables phase stepping calibration.

The issues in 6.3.2 was a lot of memory error, but I see a lot of memory related commit. 

I would say, just be patient and wait on 6.3.3 to go live for core one

rinkek, 2025-05-23:
> > I don't think 6.3.3 contains anything new related to phase stepping. I tested M1977 on 6.3.2 and got the memory error every time, so i'm pretty sure i'll get this error on 6.3.3 too.
> 
> It does. See [20bf392](https://github.com/prusa3d/Prusa-Firmware-Buddy/commit/20bf392c8fabacb0b1abcb39534ea4a633fa099e) who enables phase stepping calibration menu.
> 
> The issues in 6.3.2 was a lot of memory error, but I see a lot of memory related commit.
> 
> I would say, just be patient and wait on 6.3.3 to go live for core one

You're totally right, i accidently upped one version.. i meant 6.2.2 not having any changes.

danopernis, 2025-05-23:
I would start by waiting for actual release 😄 sources are out, binaries are not. We don't expect folks to routinely build and flash their own firmware.

Other than that, OOMs are fixed, calibration has been tweaked, but it still sometimes fails to find correction to motor currents. Maybe because motors are already good, maybe because calibration process needs more tweaks. We don't know yet. I would say code is definitely more stable than in the previous release. Instructions to bypass hidden menu are still the same. We still don't encourage regular people to try it and if they do, they should understand we don't consider this the final version (otherwise we wouldn't be hiding it from the menu).

Also, folks should really manage their expectations about the impact on sound levels and print quality. These issues can be also caused by hardware and if they are, firmware might not be able to fix them. Phase-stepping in particular just corrects the non-linearity of the motors.

Takrel, 2025-05-23:
How do i get the bypass? And how do i use it?
I dont want to compile it myself. No experience in this. So better not doing it myself.

Takrel, 2025-05-23:
Ok.Then i wait better. Mine runs on 6.3.2 i think.

andersnygaard, 2025-05-23:
I did try to run gcode M1977 via OctoPrint and had a crash and a blue screen every time.

The diff between 6.3.2 and 6.3.3 seems promising.

Will run again when the firmware is released and report on the difference.

e-basteln, 2025-05-23:
> I would start by waiting for actual release 😄 

That's what we have all been doing for months now. I think it's your turn to finally deliver. And I don't feel like laughing about it, closer to crying. 

I am still watching the calendar -- I was hoping to see this released, and see whether it gives my Core One acceptable VFA/ripple performance, before the 60-day return window closes.

blgli, 2025-05-23:
> [@danopernis](https://github.com/danopernis) I think part of the problem here is that the printer is advertised as having phase stepping. Which it does not... Also there are a ton of people having vertical artifacts on their prints currently and we were hoping for a better printer. It could very well be partly due to hardware but so far there hasn't been any good suggestions on how to mitigate/address the issue either

Well if you ask the devs the printer does have phase stepping. It’s just not tuned. Which I think is a horrible answer to the question. The XL didn’t have phase stepping and then when it did it was known to lower noise and improve the printer over all. So customers were expecting the same from this machine but we have been just told, just wait till it’s fully released for months.

Shushuda, 2025-05-23:
Yeah, I'm just gonna say advertising phase stepping, then going "oh it is enabled, we just don't tune it" is slightly insulting. Especially after making such a big deal of what it does in an XL blog post, which is further misleading. It's like advertising a touchscreen, then saying "oh we have it, we just disabled it with no way to enable". Or advertising an additional fan, then going "oh we do have a fan, it's just not plugged in". If phase stepping was advertised as doing specific things in a blog post, and later advertised as a feature in the upcoming Core One, then it's logical everyone expects the same functionality. I understand having to wait for a firmware update, but then that should've been clearly stated in the product description.

That aside, I also wish we had an alpha release to test. If anything, phase stepping will at least give people a clear answer whether their artefacts are caused by hardware or firmware. Atm we're kind of stuck in a cycle of "well maybe it's firmware, so please wait until an update" and "well maybe it's hardware, don't expect much from a firmware update". Can't even decide which is the case currently.

bkerler, 2025-05-25:
I think there is a bit of a misunderstanding. Phase stepping has always been enabled on Core One as advertised, but calibration of phase stepping hasn't been released yet because of issues regarding the 0.9 degree stepper motors (400-step motors) that cause some issues as they are way more precise. Thus the implementation of the XL (using the 1.8 degree steppers, less precise but easier to calibrate) cannot be used on the core one and has to be adapted first, meaning a lot of tests have to be made to ensure it's working as expected.

See [config](https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/818d812f954802903ea0ff39bf44376fb0b35dd2/ProjectOptions.cmake#L298-L300) and [comment](https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/818d812f954802903ea0ff39bf44376fb0b35dd2/src/persistent_stores/store_instances/config_store/defaults.hpp#L329-L331).

So please be patient and let the developers do their work to make sure the new implementation isn't rushed but the way it should be :)

e-basteln, 2025-05-25:
> If anything, phase stepping will at least give people a clear answer whether their artefacts are caused by hardware or firmware. Atm we're kind of stuck in a cycle of "well maybe it's firmware, so [...]

@Shushuda -- I think the hope is "It's hardware (namely motor imperfections), which can be compensated for by firmware (namely phase stepping with motor-specific calibration, resulting in extra-smooth motor movement)."

blgli, 2025-05-25:
https://blog.prusa3d.com/phase-stepping-how-we-busted-vibrations-and-improved-print-quality-on-the-xl-printer-with-just-a-firmware-update_94793/

Cording to this blog post, release by Prusa themselves, the main things achieved with phase stepping enabled is reduced vibrations and better print quality. My core one is the loudest printer I own and suffers from SERVE vibrations and resonance on certain moves. I fail to see how phase stepping is enabled on these machines. Comments in the firmware are just that, comments, I am not seeing real world benefits at all.

bkerler, 2025-05-25:
> > I think there is a bit of a misunderstanding. Phase stepping has always been enabled on Core One as advertised, but calibration of phase stepping hasn't been released yet
> 
> I think you need to re-read my previous comment because I address this exact point and point out why it's disingenuous.
> 
> Phase stepping being enabled "as advertised" is the same as saying a printer has input shaping enabled but setting compensation to 0, essentially making it do nothing. "But hey, it is enabled! It just doesn't do anything." Would you really say the printer has enabled input shaping in such a case? I don't think so. Now replace IS with phase stepping in these sentences.
> 
> As for waiting, that is exactly what everyone here has been doing all this time. We even offered to help generate testing data for Prusa team. We want our printers to work properly and are willing to put our own time into making this a reality.
> 
> The lack of patience from some comes from the false advertising and numerous quality and noise issues with the printer. People are just annoyed they paid for something they still have not received "as advertised". If you ask for understanding from customers towards Prusa, you also should have understanding towards the source of customers' frustration.

My previous printer was a modded and perfectly tuned MK4S that was almost unhearable. Is the Core One much louder ? Yes, totally. But this is Github issues and not a forum. If you want to rant, please feel free to rant at discord or the forum. The discussions here should stay technically, thanks :)

bkerler, 2025-05-25:
> [@bkerler](https://github.com/bkerler) can you explain what it does when it's "enabled" but not calibrated? can you show me a before/after of it being enabled and disabled?

@BlueFyre I have just began reviewing the core one code as I got my upgrade kit just few days ago, so I cannot yet tell the difference. Not sure it's really worth digging into as I know the dev's are already working hard on it for several weeks/months already and there should be a solution soon. If I get any more details, I'll let you know.

bkerler, 2025-05-25:
> Phase stepping without motor-specific calibration has no effect at all, in my understanding. It moves the motor in microsteps which are nominally equidistant (but will actually vary for non-ideal motors), same as without any phase stepping at all. No smoother motor movement or other benefits are obtained.

It can have effects, both positive or negative, depending on the default values if calibrations do not happen. I can only assume that the main issue currently is that they haven't found the ideal calculation and/or enough data to confirm the best calculation method as each additional stepper can also have side-effects but also different levels of belt tension/trapezoid effects. But enough on speculations, time to dig more into technical details and code :)

rinkek, 2025-05-26:
> > I did try to run gcode M1977 via OctoPrint and had a crash and a blue screen every time.
> > The diff between 6.3.2 and 6.3.3 seems promising.
> > Will run again when the firmware is released and report on the difference.
> 
> Can confirm the wizard ran ok on firmware 6.3.3

Did the same, calibration ran fine. 
According to the calibration i had x reduction 33%, y reduction 85%.
When printing i didn't notice much, except with 45 degree printing, like the first and second layer.
I believe the x could be better, i still hear a bit of resonance there.

andersnygaard, 2025-05-27:
> > I did try to run gcode M1977 via OctoPrint and had a crash and a blue screen every time.
> > The diff between 6.3.2 and 6.3.3 seems promising.
> > Will run again when the firmware is released and report on the difference.
> 
> Can confirm the wizard ran ok on firmware 6.3.3

I did reprint a part with noticeable VFAs after running the phase stepping wizard. In terms of noise, some of the most pronounced resonance seems to be gone. However, there are no difference in print quality. The new part is on top:

[https://imgur.com/a/Yg5LS4h](https://imgur.com/a/Yg5LS4h)

Habakuck1956, 2025-05-27:
Did you check the belt tension being symmetric in frequency? Maybe it’s also worth a try adjusting resonance frequencies above and below the recommended 85hz. On my Core one everything is clean and shiny.

[29 more replies]