prusa3d/Prusa-Firmware-Buddy #4462: [BUG]

ronperek, 2025-02-06:
### Printer model

XL Multi Tool

### Firmware version

6.2.0

### Upgrades and modifications

None

### Printing from...

USB drive

### Describe the bug

When using the printer long enough, you will run into "Puppy discovery error. No puppy found" Error code: #17513 due to puppy connection getting loose, then the red screen shows, and you need to discard the print, reconnect plugs and try again. Some form of print recovery should be available instead.

### How to reproduce

Use the printer long enough and you will eventually run into this issue. Then it will happen regularly.

### Expected behavior

I assume that when puppy connection is interrupted, the x,y,z position of the extruder is not corrupted in any way.
Why instead of showing the red screen of "throw away your xx-hour print" the printer would show some notification instructing to wiggle puppy connector or reconnect it and try to recover the print?
Guess the Prusa won't be willing to ship me or any of the affected users any upgraded hardware that will permanently solve this issue, so the next best thing would be if I can at least not waste plastic and print time every week of printing.

CloudStrife1983, 2025-02-06:
If this dont help. Replace the dwarf board and Cable. Write with support in this case.

ronperek, 2025-02-06:
Thank you. That hardware modification looks promising. I'll try it.
Change in the firmware can be made anyway. Always better to recover that to put plastic in trash.

closed 2025-04-15 by github-actions[bot]