prusa3d/Prusa-Firmware-Buddy #4463: [FEATURE REQUEST] Filament loading after layer calibration.

duedipicche, 2025-02-06:
### Printer model

MK4/MK4S/MMU3

### Describe the feature

Is it possible to have the filament loaded after the bed calibration has been done? In the MK4S+MMU3 this function exists, but if the printer is used in "single mode" or without the MMU3 the printer heats the nozzle, loads the filament and then cools the nozzle to the bed calibration temperature. This causes the filament to drip from the nozzle and the nozzle cleaning system does not work properly and the calibration is incorrect due to the presence of  not melted filament under the nozzle. The presence of filament makes the PTFE tube less flexible, falsifying the load cell reading. Thanks

me25542, 2025-02-06:
I think it would be possible, with the right start gcode

VirtualGentlemen, 2025-02-06:
this should be possible already, by modifying start sequence g code. 
It would be much nicer if it is implemented as a feature for everyone, so we don't have to touch g code :)

duedipicche, 2025-02-10:
The initial g-code of the MK4S with or without MMU3 are the same, the only difference is that the MMU3 is set,
 I think that the sequence is in the firmware and cannot be changed with the initial g-code.
 If the old version that allowed the use of the MMU in "single" or "multi" mode was restored, it would be nice. 
I also use the MMU3 to have different types of filament, but this requires that I can only choose between PETG or PLA, while if I want to print ABS or other materials I have to choose the "mono" mode with all the problems mentioned above.

closed 2025-04-21 by github-actions[bot]