prusa3d/Prusa-Firmware-Buddy #4464: [BUG]  filament management: User filaments have 'warnings' that need to be confirmed at printer

VirtualGentlemen, 2025-02-06:
### Printer model

mk4s

### Firmware version

6.2.0

### Upgrades and modifications

none

### Printing from...

PrusaConnect

### Describe the bug

'material mismatch' warning always appearing, and must be confirmed at printer. annoying.

### How to reproduce

create a custom user filament at the printer. i did one with custom temperatures for my TPU.
At the slicer, slice a model using the correct settings. My material is set to 'flex'

[image]

printer sais mismatch of material, and warning must be confirmed. I cannot set user filament profile to a material to match slicer material.

### Expected behavior

allow user filament profiles at the printer to set the material, similar list to slicer

[image]

and if both are same, no warning please!

ojrik, 2025-02-07:
Hello.
Just write the name of your own filament in the slicer, do not select from the list, but write the name of your filament.

ojrik, 2025-02-07:
[image]

[image]

VirtualGentlemen, 2025-02-07:
thank you ojrik, this has resolved the warning message.

i guess it was User error or un-intuitive UI: I did not see that a dropdown menu can be customized.

To the devs: add a button next to it, title: CUSTOM. When clicked, let a text field appear, where one can enter custom name.

closed 2025-02-07 by VirtualGentlemen

ekohlwey, 2025-05-01:
There is still an issue here if people are using a filament that is in the slicer but not in firmware. In my case it is PVA. I would like to just us a PVA profile in the slicer, but due to this bug it is impossible to configure my loaded filament in the firmware in such a way that allows selecting PVA with no warning. I could I suppose make another profile and call it something like CUSTOMPVA but this, to me, is a very dissatisfactory outcome. This should "just work" on common filaments.