prusa3d/Prusa-Firmware-Buddy #4465: [BUG] Numeric Error Accumulation Causing Layer Shifts

murk-sy, 2025-02-07:
### Printer model

Mini+

### Firmware version

6.2.0

### Upgrades and modifications

Filament sensor

### Printing from...

USB drive

### Describe the bug

This is a known issue from release notes, but I believe it is a regression as it started appearing on known good gcode.
Was not an issue on 6.2.0-alpha2, can't say for sure about 6.2.0-RC1.

On 2 different gcode files so far I've encountered consistent layer shifting, of course using gyroid (that I usually use).

Gcode B (on Mini 3)
This one was sliced recently and failed both times it was printed.
The shift is by exactly the same amount.
[image] [image]

Gcode A (on Mini 2)
This one was sliced on Prusaslicer 2.8.1 and was printed a dozen or so times on both Mini 2 and Mini 3 before the current stable version, without any layer shifting. **This is the main reason why I believe it is a regression of some sort.**
[image] [image]

### How to reproduce

Gcode in question can be provided privately to Prusa devs if needed

dolezelk, 2025-02-07:
same nasty issue on Mk4S. Wasted a lot of filament worth $20, will Prusa cover that ?

CZDanol, 2025-02-08:
The accumulation bug is not a regression. You're possibly attributing it something it didn't cause.

danopernis, 2025-02-08:
@murk-sy Sorry to hear about that. As a workaround, try using different infill if possible or downgrade the firmware. Yes, we would like to receive your G-Code for our internal testing. Did you observe this on other printers, or just MINI?

murk-sy, 2025-02-08:
@CZDanol I can't say for sure it's the issue mentioned, but considering it happens on previously working gcode, I think it's safe to say it's a regression of some sort.

@danopernis Sent an email with both gcodes to info@prusa3d.com, title "Prusa-Firmware-Buddy Issue 4465: Numeric Error Accumulation Causing Layer Shifts". Total 24MB but hopefully there won't be issues receiving it.

I've only noticed the issue on Minis, haven't had much prints on the XL with any decent amount of gyroid infill since 6.2.

After cancelling the print I *believe* there were skipped steps on both axes, but I don't remember 100% for sure.

pepa1367, 2025-02-08:
I have also experienced this issue on my MK4S fw6.2.0. In the second try the layer shift happened exactly at the same layer. [image]

CZDanol, 2025-02-09:
@murk-sy it is possible that your issue is a regression. But the numeric error accumulation is not a regression, it has been in the firware for some time, we have just identified it now.

dolezelk, 2025-02-12:
> [@dolezelk](https://github.com/dolezelk) has this part printed fine before, ideally with the same filament and gcode? Or is it the first time you have printed it? On what printer model was this printed? Can you share gcode if possible?
> 
> Consistent layer skips can be caused by different things as well, so I wanna avoid bundling in issues that aren't on gcode that worked before.

I woud like to share the gcode, but have no idea how to. When I try to attach it to this comment, it says .bgcode not allowed.
gcode is 48MB.

My printer is MK4S kit version. 
Works fine with 6.1.4, layer shifts happen just with firmware 6.2.0

dolezelk, 2025-02-12:
Failed to upload "0.4n_0.15mm_PLA_MK4S_18h54m.zip" 
might be too big, 48MB .....

mq1, 2025-02-13:
This issue also applies to adaptive cubic infill

murk-sy, 2025-02-13:
@mq1 Has your print worked fine before? Is the layer shift consistent?

If so, then I would guess the issue is triggered by any kind of wobbly perimeters too

danopernis, 2025-02-13:
@murk-sy just to keep you posted, we received both your files and one of them printed without layer shift on current development version of the firmware (there was a bit of warping in one corner but that's probably due to air drafts in the hallway where it printed). The second print is still in progress.

That being said, we are still not sure when the next release will be and if it would include this particular fix (and what else this fix breaks) I'll talk to higher-ups about the potential alpha release, but can't promise anything. For now, the only workaround is unfortunately to downgrade, which sucks because there are so many nice features in 6.2.0, mainly regarding USB reliability.

dolezelk, 2025-02-13:
> [@dolezelk](https://github.com/dolezelk) you can compress the gcode into a .zip or .gz file and then you should be able to upload it fine.

did put it here, it is big:
https://www.uschovna.cz/zasilka/RJCLXSJVXRLKPKIC-I2A/LRJ89Z949W

Today got a new Filament, flashed printer to 6.1.3 FW and now printing that gcode again. 18h print. Will share my results

dolezelk, 2025-02-14:
6.1.3 FW MK4S - using exactly the same gcode, no layer shifts. 
Problem is definatelly in the 6.2.0 firmware

danopernis, 2025-02-14:
Oh no 😦 

I am terribly sorry. Our internal tests went alright on all printers and 3d models, except for one model on MINI with gyroid infill. I will bump priority on this.

mq1, 2025-02-14:
> [@mq1](https://github.com/mq1) Has your print worked fine before? Is the layer shift consistent?
> 
> If so, then I would guess the issue is triggered by any kind of wobbly perimeters too

I was able to print correctly after downgrading to v6.2.0-alpah2.

Yeah, the layer shift seems to be consistent.

dolezelk, 2025-02-14:
Now I remember I had little layer shift with v6.2.0-RC1 on MK4S, maybe it also helps to find what code that causes this

DanielBryars, 2025-02-14:
I've never had a layer shift using 6.1.3, but after updating to 6.2 I've had several. I've rolled back to 6.1.3 earlier today and I'm now printing reprinting the failed ones. It's going well so far, no issues. However I did spend a lot of time checking the belts and tightening all the screws, so maybe it's that. I'll A/B/A test 6.2 when I've got some time, but I'm pretty sure there's something going on with layer shifting with this new release.

djweber, 2025-02-16:
Just wanted to confirm that I was also observing layer shifting on my MK4S running the 6.2.0 firmware when using fuzzy skin. Downgrading to 6.1.3 fixed the issue.

djweber, 2025-02-16:
Confirming that I was also observing layer shifting on my MK4S running the 6.2.0 firmware when using fuzzy skin. Downgrading to 6.1.3 fixed the issue - left print is 6.2.0, right is 6.1.3.

[image]

CZDanol, 2025-02-18:
Internal ticket: BFW-6795

zjohnson16, 2025-02-21:
+1

Layer shifting on a previously working gcode and a new gcode after flashing 6.2.0. The previously working gcode was a large object sliced with rectilinear infill and fuzzy skin. The new gcode was a large object sliced with gyroid infill. Rectified the layer shifts in that object by slicing with rectilinear infill instead. 

TLDR gyroid infill and fuzzy skin both triggered the layer shifts. Planning on downgrading firmwares so I can print with fuzzy skin until this is resolved. Thanks for putting in the work.

ScottWell1, 2025-02-21:
+1, this is critical and really seems like it deserves and EMERGENCY fix release -- not just waiting until the next "normal release" on the schedule.  Please light a fire under this.  :-)

CZDanol, 2025-02-21:
Yep, we are working on this

danopernis, 2025-02-21:
Thanks to all folks reporting this and taking time (and filament!) to help debugging this issue.  We believe we have a fix ready and it is undergoing last round of testing. Hopefully we will release 6.2.2 version of the firmware soon.

marcel151, 2025-02-22:
This just ruined my 12 hour print and wasted a lot of precious Prusament, wasn't aware that this is such an issue. I never experienced layer shifts before. I am using stock grid infill btw. This is critical and has to be fixed ASAP! In my opinion, this firmware never should be released, I do not agree with "This issue is not a regression, so we have chosen not to delay this release.".

MK4S
FW 6.2.1

CZDanol, 2025-02-22:
There is a regression in 6.2.0 that is causing layer shifts. I was just stating that the "layer shift caused by error accumulation" is not a regression and that this is a different issue that what was mentioned in the changelog.

You are asking us to not rush releases, but at the same urging us to fix this ASAP. We are working on this and we are veryfing that the fixes work.

We are sorry for the situation. If you want something ASAP, I suggest to downgrade to 6.1 for the moment until the 6.2.2 is out.

marcel151, 2025-02-22:
> You are asking us to not rush releases, but at the same urging us to fix this ASAP. We are working on this and we are veryfing that the fixes work.

Not to rush NEW releases. But I think it's understandable that you should prioritize fixing issues that make the printers unreliable (what you did)? Smaller prints work fine (at least for me), but bigger prints seem to fail. I was troubleshooting the whole printer (which worked totally fine until now), including using other USB etc. until I remembered that there was this passage with layer shifts in the changelog. I assumed, that gyroid infill was causing that, that's why installed it nevertheless. This shouldn't be a release FW when FW has this big problem if you ask me. Problem seems to be bigger than estimated, though.

zjohnson16, 2025-02-23:
I'm sure you already pinpointed the problem code when you made the fix. But just in case I'll add that rolling back to the alpha versions of 6.2.0 caused my gcodes to print correctly again. I didn't need to roll all the way back to 6.1.

maru-sama, 2025-03-01:
It looks like I was bitten by this bug twice with the same model. I got  a rather tremendous layer shift after around 9 hours of printing, each time at the same layer. Reading through the comments I see that the infill (gyorid) seems to be the main issue. Until a new firmware is released, is it one correct that switching to a different infill like rectiliniar fixes the issue? I do not want to have a failed print again after 10 hours.

murk-sy, 2025-03-01:
It's not guaranteed because it seems to depend on just all geometry - if you have a wavy outside of the model like in https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4465#issuecomment-2653609700 you may still run into the same problem. Safest workaround right now is to downgrade to 6.1.3 version.

I haven't had issues with the models I posted initially after using cubic infill but the geometry is otherwise pretty simple.

maru-sama, 2025-03-01:
> It's not guaranteed because it seems to depend on just all geometry - if you have a wavy outside of the model like in [#4465 (comment)](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4465#issuecomment-2653609700) you may still run into the same problem. Safest workaround right now is to downgrade to 6.1.3 version.
> 
> I haven't had issues with the models I posted initially after using cubic infill but the geometry is otherwise pretty simple.

Hi, thanks for the quick reply, would switching back to the 6.2.0-alpha2 version also work?

maru-sama, 2025-03-01:
Just to be on the safe side I reverted back to 6.1.2 (the latest 6.1.x for the MINI) and restarted the print with the same gcode file. I will provide an update in around 10 hours.....

marcel151, 2025-03-01:
> would switching back to the 6.2.0-alpha2 version also work?
Yes, that will also work. I still think this was a mistake to release a production firmware with a known critical bug in the first place. In defence to Prusa, this bug seemed to be more critical as they thought at first. Hoping for a fix next week.

maru-sama, 2025-03-02:
Hello, the print just passed the layer it always happened in the past and there is no layer shift. Furthermore the print also seems to be more silent, when doing the infill.

VSCustoms, 2025-03-02:
So anything below 6.2.0 Alpha causes BSOD with MMU3. Anything from 6.2.0 Alpha causes layer shit with fuzzy skin. Change only fuzzy skin and everything is fine. 

All in MK4S…..I can’t run below 6.2.0 any because I’m running a 12 colour MMU and only 6.2.0 likes past 5 colours.

maru-sama, 2025-03-03:
> So anything below 6.2.0 Alpha causes BSOD with MMU3. Anything from 6.2.0 Alpha causes layer shit with fuzzy skin. Change only fuzzy skin and everything is fine.
> 
> All in MK4S…..I can’t run below 6.2.0 any because I’m running a 12 colour MMU and only 6.2.0 likes past 5 colours.

From what I understand 6.2.0-alpha2 should work as well.

VSCustoms, 2025-03-03:
> > So anything below 6.2.0 Alpha causes BSOD with MMU3. Anything from 6.2.0 Alpha causes layer shift with fuzzy skin. Change only fuzzy skin and everything is fine.
> > All in MK4S…..I can’t run below 6.2.0 any because I’m running a 12 colour MMU and only 6.2.0 likes past 5 colours.
> 
> From what I understand 6.2.0-alpha2 should work as well.

6.2.0 Alpha2 is what I currently tested and same issue. 
 Rectilinear infill, fuzzy skin set to .1 T and .2 D and it layer shifts. 

I turn off fuzzy skin and there’s no issues.

zjohnson16, 2025-03-03:
Hmm 6.2.0 alpha2 is working for me where 6.2.0 did not. My print appears to be smaller than yours however and I used 0.2PD 0.2T, so your print probably has a lot more points. That might be the culprit. 

Are the layer shifts happening consistently on the same layers each time though? Because that was the big giveaway for me that this was a software issue and not a hardware issue. Rectilinear infill can cause layer shifting by itself since the nozzle crosses previously laid infill.

VSCustoms, 2025-03-03:
> Hmm 6.2.0 alpha2 is working for me where 6.2.0 did not. My print appears to be smaller than yours however and I used 0.2PD 0.2T, so your print probably has a lot more points. That might be the culprit.
> 
> Are the layer shifts happening consistently on the same layers each time though? Because that was the big giveaway for me that this was a software issue and not a hardware issue. Rectilinear infill can cause layer shifting by itself since the nozzle crosses previously laid infill. When the fuzzy skin makes a layer take longer it could create some weird warping issue with the infill.

See I would be on the same trail of infill being the issue but all these zero infill prints were done with all the firmwares and the only one that worked was fuzzy skin off. 
 And I man my MK3S+ and MK2.5S have always printed with my fuzzy settings. 

[image]

danopernis, 2025-03-05:
Hi folks, we have released 6.2.2 firmware which should fix this issue.

murk-sy, 2025-03-05:
I'll run gcode B as in the first comment and see if all goes well - not really expecting surprises though. If so, I'm closing the issue tomorrow, I'll reopen it if someone else still reproduces it afterwards.

VSCustoms, 2025-03-06:
> Hi folks, we have released 6.2.2 firmware which should fix this issue.

Ran 6.2.2 on one the same Gcode fuzzy skin that shifted before and it printed perfect this time. Just printing a second one now.

murk-sy, 2025-03-06:
Resolved with 6.2.2, closing

closed 2025-03-06 by murk-sy