prusa3d/Prusa-Firmware-Buddy #4468: [BUG]

a-pushkin, 2025-02-08:
### Printer model

XL 5 tool

### Firmware version

6.2.0

### Upgrades and modifications

None

### Printing from...

USB drive

### Describe the bug

While printer was turned on an idle it went into a panic screen referencing tmc_util.cpp:1224
Power cycling the printer resulted in same panic screen upon boot. During the boot lights on only first 2 toolheads out of 5 would turn on.
Eventually the problem went away by itself.
Crash dump can be downloaded from https://1drv.ms/u/s!AgDo7289PrYqlZZ3dIPIJABGCWYb1g?e=9gVNaT (couldn't attach it to a bug report directly)

CZDanol, 2025-02-11:
Hello, thank you for your report. Internal ticket: BFW-6759

CZDanol, 2025-03-22:
You are saying that you have the same problem. So, you are also getting a "tmc_util.cpp:1224" bsod?

Prusa-Fan, 2025-03-24:
Yes. Bluescreen with "tmc_util.cpp:1224".
But in the meantime, I found a way to avoid the crashes and start printing.
I found out, that the printer only crashes during booting, when the USB-Stick is installed.
After removing the USB-Stick booting works without any problem. After booting, I can install the USB-Stick and prints start normaly.
I reformatted the Stick (FAT32), but the result was the same: crash with, normal booting without stick.

But to my surprise, since yesterday booting with inserted USB-Stick suddenly works again.
So I have no idea, what might be the real reason for the problem.

d0mse, 2025-04-09:
> Since yesterday, I have the same problem! At first, I only had sporadic crashes, but now the printer won't start at all. Is there any explanation for this problem?
> 
> For some time now, I've also been experiencing frequent crashes when starting my prints with the error message "Homing Error Z." However, all tests are OK, and the error occurs even when the nozzle isn't even close to the print bed. Could the errors be related? After a reset, the printer always starts without problems.

I have the exact same Problem. After reset the Printer makes the tmc_util.cpp error. The only fix is to reboot without the usb.

Prusa-Fan, 2025-04-13:
Concerning the problem with the "Homing Error Z", I got the following information from the Prusa-Support:
"If the load cells are OK when you tap the nozzle, it would be a good idea to try moving the toolhead once it's picked up and see how the value on the load cell changes, as sometimes the Dwarf cable itself can affect the load cell.
You can find information about the load cell value by going to Menu - Info - Sensor Info - and then scrolling down until you see the load cell value changing.
However, the toolhead must first be picked up for the load cell value to be displayed."

Prusa-Fan, 2025-04-14:
After installation of XL_firmware_6.2.4-RC1 the bsod is gopne.
Problem is solved!
Thanks!

microtherion, 2025-06-22:
I've seen this 3x today with 6.2.5