prusa3d/Prusa-Firmware-Buddy #4469: [BUG] Opening doors causes crash

greggjaskiewicz, 2025-02-09:
### Printer model

CoreOne

### Firmware version

6.3.0 

### Upgrades and modifications

Noe

### Printing from...

USB Drive 

### Describe the bug

Opening doors at some points causes either crash (probably triggering watch-dog, the way it behaves). 
I found this happening either during initial warm up/bed mesh, or during final descent when print is done - and I'm eager to open the door! 

Or on other occasions (probably should be separate bug) - throws UI dialog in the middle of another event. Picture running out of filament, and being prompted to confirm that correct colour has been observed. I want to then open the door and pinch away the excess - but dialog on the screen obstructs the new-filament query dialog. It's almost like door opening event has just been a last minute slap on top of the firmware. 

### How to reproduce

run print, and open the door either before print starts properly or when it finishes and bed descends down Z - to lower position. 

### Expected behavior

Opening the door shouldn't cause everything to crash or other higher priority (such as filament reload) dialogs should not give way to it. 

https://github.com/user-attachments/assets/088b0164-5d0d-4bc1-9eb1-359b4faf47ee

CZDanol, 2025-02-10:
Hello, thank you for reporting. The printer restart at the end should be fixed in 6.3.1. The one at the beginning will probably take a bit longer. Internal ticket: BFW-6705

thijstriemstra, 2025-04-03:
nobody at prusa thought about ever testing something so obvious blows my mind. lots of these gems in bug tracker.

danopernis, 2025-04-03:
@greggjaskiewicz 6.3.1 was just released and contains fixes to the issues you were having.

@thijstriemstra thank you for your valuable insights.

Avarios, 2025-04-13:
@danopernis Printer still crashes when you open the door during print

MCraft555, 2026-09-14:
I got a similar issue that the firmware crashes when opening the door after pressing retry for the bed leveling. Firmware version 6.8.1+16182.

DreitTheDragon, 2026-10-04:
Just wandering around, I noticed this issue and want to share one _thought_ I had - couldn't it be caused by wrong data in non-volatile memory from older firmware releases? I had weird issue after upgrading to 6.8.1, printer worked well with no issues, printed few bigger parts, was turned on and off several times.
Few days later I wanted to run some calibration, so I started it in menu and printer (Core One L) suddenly says it doesn't have accelerometer connected. Hmm, that's weird, it worked perfectly last time. I tried it few times, it always failed and complained about accelerometer being disconnected. So I tried to load default values, that didn't work either. In the end printer simply refused to start any print because it's not calibrated. Great!

Only thing which really helped in the end was to do factory reset, it started setup wizard, finished everything with literally no issues or hiccups (suddenly accelerometer was connected and working) and it runs perfectly now. That experience brings up question on my side - can't new firmware accidentally read some old data from memory and act differently, compared for fresh install? Maybe reading different addresses/offsets since INDX and other stuff was added later? And if not, why would factory reset clear my calibration issue?