prusa3d/Prusa-Firmware-Buddy #4470: [BUG] Stamping/Ramming before each layer change

Bladeson01, 2025-02-09:
### Printer model

MK3.5 with MMU3

### Firmware version

6.2.0 (stable release)

### Upgrades and modifications

custom extruder: VZ-Hextrudort with stock hotend

### Printing from...

Prusalink, Prusa Connect

### Describe the bug

Printer is ramming/stamping at the beginning and the end of the purge line and also before each layer change, resulting in filament blobs.

### How to reproduce

Flash to previous firmware 6.1.3 fixed the issue.

Bladeson01, 2025-02-21:
No chages with FW 6.2.1.

CZDanol, 2025-02-25:
Hello, could you please share some gcode that exhibits it?

Also, regarding the custom extruder - does it have some non-standard parameters for the extruder motor?

Bladeson01, 2025-02-25:
Hi @CZDanol , sure yes, here is the some out of the gcode I have used in the videos.
```
; Extrude purge line
G92 E0
G1 E2 F2400
G0 E7 X15 Z0.2 F500
G0 X105 E36 F500
G0 X115 E4 F650
M73 P1 R30
M73 Q1 S38
G0 X125 E4 F800
G0 X128 Z0.05 F8000
G0 X131 Z0.2 F8000     ; <--- between here ↓
G92 E0
M221 S100
G21
G90
M83
T4
M572 S0.036
M142 S36
M107
;LAYER_CHANGE
;Z:0.2
;HEIGHT:0.2
M486 S0
G1 E-.7 F2100
G1 Z.8 F720             ; <--- to here it is ramming
G1 X123.911 Y108.423 F18000    ; moving to print start point..
```

also (0:58 in video):
```
G1 X118.51 Y105.127 E.01851
G3 X118.411 Y105.643 I-1.799 J-.076 E.02405
G3 X118.333 Y105.726 I-.094 J-.01 E.00558
G1 X118.018 Y105.412 E.02028
M204 P4000
M106 S114.75
;LAYER_CHANGE
;Z:0.4
;HEIGHT:0.2
;BEFORE_LAYER_CHANGE
G92 E0.0
;0.4
M201 X3999.89 Y3999.89
G1 E-.7 F2100
G1 X118.018 Y105.412 Z.2 F18000
G1 X99.259 Y105.494 Z.527
;AFTER_LAYER_CHANGE
;0.4
M74 W0.103992
G1 X99.259 Y105.494
G1 Z.4 F720
G1 E.7 F1500
```

1:25 in Video:
```
G1 X135.527 Y102.362 E.11435
G3 X136.105 Y103.026 I-8.26 J7.778 E.0299
G1 X136.614 Y105.156 E.07437
M106 S114.75
M204 P4000
M106 S229.5
;LAYER_CHANGE
;Z:0.6
;HEIGHT:0.2
;BEFORE_LAYER_CHANGE
G92 E0.0
;0.6
M201 X3999.74 Y3999.74
G1 E-.7 F2100
G1 X136.614 Y105.156 Z.4 F18000
G1 X110.096 Y105.396 Z.863
G1 X106.169 Y105.431 Z.923
G1 X102.241 Y105.466 Z.966
G1 X99.164 Y105.494 Z.987
;AFTER_LAYER_CHANGE
;0.6
M74 W0.240788
G1 X99.164 Y105.494
G1 Z.6 F720
G1 E.7 F1500
```
For the extruder gear, I set the axis_steps_per_unit via `M92 E1465.45 ;Hexdrudort` in the start gcode in Prusaslicer and also in the experimental settings of the printer to to a maximum of 1000 . Apart from the two settings, nothing else is changed for the extruder.

CZDanol, 2025-02-26:
Would you please share the whole gcode file? The point is that we would try to set your settings and print your gcode and try replicating the problem first in the most similar conditions possible to verify that we are able to replicate the problem. Then, we would get into analyzing and simplifying