prusa3d/Prusa-Firmware-Buddy #4474: [BUG] Inescapable UI while unloading filament after filament runout; requires printer reset

neuron-whisperer, 2025-02-11:
### Printer model

MK4S+

### Firmware version

6.1.4

### Upgrades and modifications

None

### Printing from...

PrusaLink

### Describe the bug

The Prusa firmware detects filament runout via the filament sensor. Upon detecting runout, the printer beeps to alert the user, walks the user through the filament reloading process, and resumes the print.

Typically, this process involves backing out the filament so that the user can remove it and insert a replacement. However, occasionally, the back-out process does not work and the final length of filament is not ejected from the extruder. However, this is often not a problem because, in most cases, the spent filament spool will be replaced with a fresh spool of the exact same filament. If the printer could resume the loading process, the new filament will push the final length of old filament into the extruder and will then directly follow it. The only issue in this case is that the extruder gear sometimes skips a few notches at the transition point, but it's usually a brief occurrence that resolves by itself without any problem.

However, when this circumstance arises, the GUI is problematic.

The printer instructs the user to remove the ejected length of filament, and that instruction is presented until the filament sensor detects the removal of the filament. However, because no filament has been ejected, the user cannot easily remove it. They might open the idler and pry out the filament from between the extruder gear, but this isn't a delicate process, and the UI features no such instructions.

In this predicament, the GUI offers no other options. The user cannot skip the removal step; the user cannot cancel the filament swap or the print; and the user cannot return to the main menu or take any other action. The GUI is effectively stuck until (1) the user finds a way to remove the filament or (2) the user resets or power-cycles the printer. After the reset or power-cycling, the printer works fine and believes that no filament is loaded, so it is easy to load the new filament (pushing the old filament out of the nozzle as described above). However, this of course ruins any in-progress print and prevents resuming the print in place.

### How to reproduce

Start a print. Let the filament run out. Observe the GUI step to remove the filament length from the extruder, and the absence of any option for progressing if the filament cannot be removed - the only option is to reset (or power-cycle) the printer.

### Expected behavior

The GUI should provide some fallback options in case the filament removal cannot be performed. Ideally, "Skip" and perhaps "Retry Eject" options would be included, but at the very least, "Cancel Print" should be provided as an option.

dolezelk, 2025-02-14:
This happenned me a few time already. Sometimes it working fine, someting not.

neuron-whisperer, 2025-03-12:
An update:

This bug is still occurring, and appears to occur in other circumstances as well, including when the filament was properly ejected. It's weird - the printer asks the user to pull out the filament plug, and once the ejected filament plug is removed, the printer instructs the user to remove the filament. It's the same display with the same problem of inescapability even though no filament remains in the hotend.

After a reset, the printer indicates that no filament is loaded. Filament can be loaded normally at this point, which includes the usual steps of detecting no filament, asking the user to insert it, detecting the presence of filament, and heating up the hotend 
to load it. These steps are notable because the error leading to the UI hang *looks* like the filament sensor is returning a false positive (indicating that filament is still inserted when it isn't), but after a reset, the filament sensor is working fine.

Speculation: There appears to be either an inconsistency with the filament sensor - e.g., different thresholds are being used during normal filament loading/unloading and removing the filament after runout, where the sensitivity and false-positive rate are both higher. Alternatively, this could just be a logic bug that causes the printer to operate as though filament is loaded when the filament sensor says that it has been removed. I will try to record video of this occurrence and upload it to document the problem.

neuron-whisperer, 2025-04-02:
This happened again today, ruining a two-hour print that was at the 90% mark. This is getting really tiresome.

My printer was beeping due to filament runout. The filament was properly ejected out the top of the extruder, so I removed it and pressed "Yes" to the "Was the filament properly ejected?" prompt.

The printer then long-beeped and showed this:

[image]

Notice the display has no options whatsoever. Also notice that there is no filament in the extruder.

And just to be completely certain, I opened the idler...

[image]

...all clear.

I tried tapping the display, turning the dial, and pressing the button. No response.

I closed up the idler and tried inserting and removing the filament multiple times. No response.

So I sighed and hit the Reset button, and my Prusa power-cycled and then showed me the main display. No option to continue print, of course, so I just threw it away.

I selected the option to load filament, and my printer prompted me to insert filament - clearly, it didn't think that filament was inserted. I then inserted filament and it responded with the "Filament inserted, press button to continue" prompt, and then loaded filament like normal. So, clearly, the filament sensor is working perfectly.

Prusa, **please** do something about this. This is an obvious problem with this step of the printing process, it occurs frequently, it is amenable to multiple solutions (fixing filament detection and/or fixing the dialog to include some basic options), and it is ruining prints for no valid reason. Can you please spend some time on this issue?

neuron-whisperer, 2025-04-04:
I have spent some time looking into this issue, and I found what I believe to be a bug in the firmware.

I tracked down the logic for this screen to a function in pause.cpp (excerpted here):

```
void Pause::filament_not_in_fs_process([[maybe_unused]] Response response) {
    ...
    if (!FSensors_instance().has_filament_surely(LogicalFilamentSensor::autoload)) {
    ...
```

In essence, this function goes like this: If the user responded "Yes" to "has the filament been ejected," check with the filament sensor to make sure that it isn't detecting filament. Wait for the filament sensor to respond "no filament" for a full second before continuing with the process.

So what's the problem? It's this part:

`has_filament_surely(LogicalFilamentSensor::autoload)`

Looking through the entire firmware codebase, every other time that the printer needs to check the filament sensor, it does this:

`has_filament_surely(LogicalFilamentSensor::extruder)`

The **only** place in the entire Buddy firmware that checks the "autoload" LogicalFilamentSensor instead of the "extruder" LogicalFilamentSensor is the filament_not_in_fs_process function. That's why it behaves so oddly.

So what is "LogicalFilamentSensor:autoload?" This is where things get weird. Looking in the file filament_sensor_types.hpp, I find this:

```
enum class LogicalFilamentSensor : uint8_t {
    /// Filament sensor on the current extruder
    extruder,
    /// Side sensor for the current extruder
    /// MK4+MMU: MMU sensor | XL: current side sensor | OTHER: none
    side,
    /// The first runout filament sensor - the one further from the extruder
    /// XL: side sensor | MK4+MMU: MMU sensor | OTHER: extruder sensor
    primary_runout,
    /// The second runout filament sensor - the one closer to the extruder
    /// XL,MK4+MMU: extruder sensor | OTHER: none
    secondary_runout,
    /// Filament sensor that is used to detect autoload
    autoload,
};

static constexpr size_t logical_filament_sensor_count = 5;
```

Okay, there's an "extruder" sensor, a few other sensors for the XL and MMU, and then... this "autoload" sensor, with no meaningful description. The rest of the firmware treats this sensor just like the extruder sensor, by polling it to detect its state and raise events and such.

The crux of the issue, I think, is that I don't believe that the MK4/4s *has* an "autoload" sensor separate from the main extruder filament sensor. If this sensor doesn't exist, then obviously its state never changes, it never gets any events, and the filament_not_in_fs_process function that is awaiting a "no filament detected" event on this sensor will wait forever - exactly what I'm encountering.

The solution here is very simple: just change has_filament_surely(LogicalFilamentSensor::autoload) in that function to has_filament_surely(LogicalFilamentSensor::extruder) to be consistent with the rest of the code. But before I submit that change as a git commit, I want to see what people know about an "autoload" filament sensor in the context of the MK4/4S.

bkerler, 2025-04-13:
Digging into this:
https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/master/src/marlin_stubs/pause/pause.cpp#L893-L909

There are two options when a filament is stuck:
1. Use human interaction, which is the case for :
    https://github.com/prusa3d/Prusa-Firmware-Buddy/Prusa-Firmware-Buddy/blob/master/ProjectOptions.cmake#L314
    This sets the state LoadState::unloaded_ask ... which will then finally call LoadState::filament_not_in_fs
2. No human interaction, based on a timeout in "runout_timer_ms", directly sets the state LoadState::filament_not_in_fs

This issue seems to be that in LoadState::unloaded_ask, the runout timer isn't called in advance:
"runout_timer_ms = ticks_ms();"

https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/master/src/marlin_stubs/pause/pause.cpp#L917

bkerler, 2025-04-13:
I now write a fix for this issue : #4599

neuron-whisperer, 2025-04-29:
> I seem to understand that the observation of the issue originated from a partially jammed filament that failed to be unloaded due to an irregular/bulky tip at the end of the filament, which wouldn't pass through the upper extruder's hole.

What? No, this is totally incorrect.

The filament was completely ejected. I even posted photos above, including the photo of the idler that is completely clear. I don't understand what part of my report could have led to this interpretation. Can you explain?

The problem, as I indicated above in great detail, is that after completely and successfully ejecting the filament, the firmware appears to check the status of a sensor that does not exist on the MK4S - the "autoload" filament sensor. And this looks like the only instance anywhere in the firmware where this nonexistent sensor is checked with the has_filament_surely method; every other invocation of has_filament_surely throughout the firmware checks the "extruder" filament sensor.

Prusa-Support, 2025-04-29:
> The filament was completely ejected. I even posted photos above, including the photo of the idler that is completely clear. I don't understand what part of my report could have led to this interpretation. Can you explain?

That's exactly what I'm talking about.
It should be addressed by the internal ticket [BFW-5545].

> In essence, this function goes like this: If the user responded "Yes" to "has the filament been ejected," check with the filament sensor to make sure that it isn't detecting filament. Wait for the filament sensor to respond "no filament" for a full second before continuing with the process.

The described behaviour is supposedly the expected behavior - not a bug?
_In case of this occurrence, clearing the filament sensor would be necessary, as per the previous comment - https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4474#issuecomment-2839439305._

_**Additional notes.
The internal ticket BFW-5545 is actually already fulfilled in firmware version [6.3.1](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.3.1), which is specific to Prusa Core One.
Prusa CO firmware will be merged with Prusa MK4/MK4S/XL firmware in the next major firmware release.**

It seems safe to close the issue as a duplicate in either case but please wait for the next major firmware release - probably 6.4.0 (?) / no official information available - and then you can let us know if you are satisfied with the "current" solution (implemented but yet to be seen on Prusa MK4/MK4S/XL).

Michele Moramarco
Prusa Research

Prusa-Support, 2025-05-08:
Sounds like a plan!
Thanks for your understanding.

Feel absolutely free to open a new issue but, _ideally, comment here below_.
If necessary, the same issue can be reopened and continued.

Michele Moramarco
Prusa Research

closed 2025-05-08 by Prusa-Support