prusa3d/Prusa-Firmware-Buddy #4475: Wipe tower Cancel

dsjove, 2025-02-12:
### Printer model

XL

### Describe the enhancement

Sometime it is the wipe tower that fails, warps, and pops off the bed. I would like to be able to cancel the wipe tower and continue print. I don't expect ideal results, but the end result is often acceptable with some post processing.

When I let the print continue, the loose extrusions eventually ruins the main print.

### Expected functionality

When I go to 'tune', I see 'wipe tower' in the list and can cancel it.

dsjove, 2025-02-12:
I think this is a duplicate of
https://github.com/prusa3d/PrusaSlicer/issues/11615

danopernis, 2025-02-13:
Correct, this would require Slicer support to work. Thanks for finding the relevant issue. Closing this one.

closed 2025-02-13 by danopernis