prusa3d/Prusa-Firmware-Buddy #4478: [ENHANCEMENT] Minor: Log closing of log file to log file just as it does when opening the file.

ulab, 2025-02-14:
### Printer model

All

### Describe the enhancement

If `file_log_enable()` writes `Started logging to file: %s` to the log file, `file_log_disable()` should write `Stopped logging to file: %s` too. :-)

danopernis, 2025-02-14:
Yeah, I agree it is inconsistent. May I suggest, even better would be to stop logging the start of logging. Each string/code we add to firmware costs precious FLASH, especially on MINI, so I would like to keep it minimal if I can. Patches welcome 🙂

ulab, 2025-02-14:
I'm fine with removing it, it doesn't really help a lot anyway?

That said, my coding knowledge is not good enough to not break things - especially since I can't just test it easily.

ulab, 2025-02-14:
That said, it looks like the string [kind of exists already](https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/310b7f890f09c1318e264c46aed80e7ecf94dd93/src/logging/log_dest_file.cpp#L136). It just doesn't log it to the file?

danopernis, 2025-02-14:
Yeah no worries, not everybody has to be a coder. Thanks for reporting anyway, I left the "good first issue" label so perhaps somebody will grab it. If nobody does, we might implement it, but I like to leave tasks such as this to encourage community  contributions.

As for logging the disabling of the log, good catch. Either that's a race condition and log gets disabled before actually logging stuff, or it is deliberate mechanism to not log to the same log that is being disabled. I can't remember right now.

CZDanol, 2025-02-15:
I would argue that it is consistent and makes sense.

1. The log "Started logging" is emitted after we start logging. It makes sense that this gets captured in the file.
2. The log "Stopped logging" is emitted when we end logging. It makes sense that this does not get captured.

ulab, 2025-02-15:
While I agree that it is minor, I'd like to make the case that having a proper "end" marking in the log can be important to decide whether the printer crashed or the logging was ended properly.

It can also help a support member to see if he received a partial file.

CZDanol, 2025-02-16:
Ok, those are very good points

CZDanol, 2025-02-17:
Internal ticket: BFW-6789

bkerler, 2025-04-13:
I'd say as this was already merged, the issue can be closed ? :)

ulab, 2025-04-13:
If it's fixed, I agree. 

Is it already included in one of the stable releases though? I don't see the final log message in the Core One 6.3.1+10362 yet.

bkerler, 2025-04-13:
Good question :) I think it will be included in the upcoming 6.4 release.

ulab, 2025-04-14:
I'll be back if it isn't ;-)

closed 2025-04-14 by ulab