prusa3d/Prusa-Firmware-Buddy #4479: [BUG] Core One has no CHAMBER_TEMPERATURE capability and does not report C temperature

ulab, 2025-02-14:
### Printer model

Core One

### Firmware version

6.3.0+10073

### Upgrades and modifications

None

### Printing from...

does not apply

### Describe the bug

The Core One reports it has no CHAMBER_TEMPERATURE capability on `M115`:

```
Send: M115
FIRMWARE_NAME:Prusa-Firmware-Buddy 6.3.0+10073 (Github) SOURCE_CODE_URL:https://github.com/prusa3d/Prusa-Firmware-Buddy PROTOCOL_VERSION:1.0 MACHINE_TYPE:Prusa-COREONE EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
                                                                                                                                                                                                                                  Cap:SERIAL_XON_XOFF:0
[...]
Cap:CHAMBER_TEMPERATURE:0
ok
```

It also does not report the chamber temperature (`C`) on `M105` or `M155`?
```
T:26.50/0.00 B:59.83/90.00 X:26.98/36.00 A:40.91/0.00 @:0 B@:255 HBR@:0
```

### How to reproduce

Send the above GCodes `M115`, `M105` and `M155 ...` over the serial interface.

### Expected behavior

`CHAMBER_TEMPERATURE` should show `1` and the temperature reports should have a `C` with the chamber temperatures.

ulab, 2025-02-14:
I am guessing the cause for this is that Marlin [expects active heating for the chamber](https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/310b7f890f09c1318e264c46aed80e7ecf94dd93/lib/Marlin/Marlin/src/gcode/host/M115.cpp#L169)?

`HAS_TEMP_CHAMBER` instead of `HAS_HEATED_CHAMBER` might be enough, but I am not sure if changing the capabilities this way will mess with something else.

But then I am not sure why it doesn't print `C` on `M105` or `M155`. The code [looks like it should](https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/310b7f890f09c1318e264c46aed80e7ecf94dd93/lib/Marlin/Marlin/src/module/temperature.cpp#L3726-L3728).

CZDanol, 2025-02-15:
Oh I was wondering when someone would come up with this :)

Yes, we didn't implement the chamber temperature through the standard Marlin structures, because Marlin code is hell, plus it only considers heated enclosures, and the API overall was not powerful enough for our needs.

This resulted in our chamber code to be actually sane, but we are left with having to hack back some of the OG Marlin support for the Octo guys.

ulab, 2025-02-15:
I was mostly trying to get a log of the chamber temperature depending on different bed temps and to see how well the cooling works. :)

But I guess the Octo guys might want these too, yes.

CZDanol, 2025-02-17:
Created internal ticket BFW-6788

bkerler, 2025-02-17:
I just pushed a PR to make C1 great again :)

CZDanol, 2025-02-20:
We will try to finish it up internally

thijstriemstra, 2025-04-03:
> Marlin code is hell

ouch! prusa code isn't? 🤦🏻‍♂️

varukasalt, 2025-04-20:
> I was mostly trying to get a log of the chamber temperature over time depending on different bed temps and to see how well the cooling works. :)
> 
> But I guess the Octo guys might want these too, yes.

Octo guy here. Yes we very much want this. Very much.

CZDanol, 2025-04-22:
@bkerler has implemented this in #4484 . Should get out with FW 6.4.

ulab, 2025-04-22:
Closing as resolved, but still watching...! ;-)

ulab, 2025-05-28:
I can't currently test it, but this seems to be wrong anyway?

> Recv: T:230.00/230.00 B:60.02/60.00 X:37.45/36.00 A:48.65/0.00 @:137 B@:73 C@:31.61 HBR@:255

`C@:` should be "chamber power" (which doesn't make sense since the C1 doesn't have an extra chamber heater), but it seems to show the chamber temperature `C:` which is missing from the output.

bkerler, 2025-05-28:
Ok, will prepare a PR for that

fotrick, 2025-12-25:
@3d-gussner you marked it as solved but it's not - the PR #4671 is not approved yet and is not part of the 6.4.0 firmware.

jneilliii, 2026-03-10:
Current source code in the default branch is still showing that this is being reported as chamber power `C@:` instead or chamber temperature `C:`. Because of that this issue is not resolved and therefore shouldn't be closed. Are there any labels that can be applied to these issues so stale bot doesn't try to close stuff automatically that apparently is requested by several people? 

Based on the code here, I don't think this should be an if elif conditional logic, rather it should be two ifs, because as it stands the temperature report will be reporting incorrectly temperature (should be in the format of `C: <current>/target>`) as power (`C@: <power value>`). 

https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/b91eeda0c16a9931126ea065f2fa2bcc8a983b8d/lib/Marlin/Marlin/src/module/temperature.cpp#L3024-L3029

bkerler, 2026-03-24:
Hah, indeed ! Let's hope we can get it pushed through very soon.

BlueFyre, 2026-05-25:
Not stale.

angrygoats, 2026-08-31:
I would be interested in having chamber temperature reported so that Octoprint can track chamber temp through a print. Right now I have to put a secondary sensor in the chamber and while this is a little more flexible, I’d prefer it reported from firmware.

mvadu, 2026-11-02:
We are still waiting for Prusa use the canonical C: rather than C@: