prusa3d/Prusa-Firmware-Buddy #4480: [ENHANCEMENT] Allow changing the filtration setting for predefined filament profiles

unclej84, 2025-02-16:
### Printer model

MK4S

### Describe the enhancement

It would be nice if the user could change the filtration setting of the predefined filament profiles on the printer.
At the moment filtration is only enabled for some profiles but it would be nice to have it for all and the 8 available user settings are not enough when adding copies for CF variants of the available filaments

### Expected functionality

Toggle for filtration is active also for predefined profiles and can be changed by user if needed. The setting should be stored.

CZDanol, 2025-02-17:
Hello, preset filaments are "preset" for a reason - we want to keep those standardized. At the moment, we do not plan on making them customizable.

However we could discuss some other ways to potentially resolve your issue:

1. Maybe we could allow for more user filaments?
2. Maybe we could give an option to enable filtration for any print?
3. Maybe we could add a gcode "override" that you could insert into your slicer filament profile that would force filtration?

unclej84, 2025-02-22:
Any of this potential ways sounds good to me - looking forward to what might come.

zeus86, 2025-04-23:
really noone wants dozen copies of default profiles just to change one setting - be it in the printer or in the slicer. while maintaining a "good default" makes sense, some settings should be able to be overridden be the user regardless of anything else (e.G: you wouldn't require different profiles depending on the lighting, or the AC input voltage or the filament color). If you have some kind of filter behind your core one (like me), some filtration is always active, as long as the chamber-fans are running (so mostly during the print). However, what is missing is to allow the user to always enable post-print-filtration, be it through overrides or filament-settings. All thermoplatics emit particles when printing, so enabling post-print-filtration when the user desires so, is a reasonable way of handling this. For me it totally sucks, to have filtration behind my core one, but the printer does not let me do filtration, unless i copy each and every standard profile, just to change one dumb setting that has nothing to do with anything during printing (and in this case i would agree, that standard-profiles should take over here).

**If the user chooses to ALWAYS have post-print-filtration active, because the user cares about his/her health, then for the love of the flying spaghetti-monster: PLEASE LET THE USER DO SO! (especially because it does not affect the print-quality whatsoever!)**

CZDanol, 2025-04-24:
I don't know, sorry.

zeus86, 2025-11-24:
prusa really should build a bot, that keeps tickets open, when there is a connected internal ticket...