prusa3d/Prusa-Firmware-Buddy #4482: [FEATURE REQUEST] XL MCU cooling fan firmware support

JohnHP7550A, 2025-02-17:
### Printer model

XL Single/Multi Tool

### Describe the feature

In some specific use cases the MCU on the XL tends to overheat. The current solution is to power a fan from the 24V rail, so the fan runs all the time, even when the printer is idling. 
It would be useful to use the "FAN2" connector on the Sandwich board and have a controlled fan based on the MCU Temperature. Maybe the implementation of the gui functionally could be similar to the enclosure. Add a menu item to enable/disable active MCU cooling and let the firmware test the fan.
Additionally it would be great to set to MCU temperature values. A lower threshold below the fan is switched off and an upper threshold above the fan runs at 100%. Like fancontrol on linux. 

I know this is a specific feature, but it would improve and simplify the current cooling solutions. 
Thank your very much for your consideration.

### Expected functionality

MCU cooling fan connected to the "FAN2" connector on the Sandwich Board.
Menu item to enable/disable active MCU cooling.
Fan controlled based on MCU Temperature.
Menu item to set lower and upper threshold temperatures to set the fan curve.

Tomcus, 2025-02-19:
Hi this is a cool idea (pun intended). I will try to talk with someone internally about it. The FAN2 connector has only 5V connected, but that shouldn't be a problem.

JohnHP7550A, 2025-02-19:
Hello, thank you very much. 
Yes there are enough 5 V fans out there. What are the restrictions on power draw in relation to the PCB? I found a suitable fan with a maximum power consumption of 1.3 W.
Best Regards

Tomcus, 2025-02-19:
I am no 100%, the schema sais 5V and I am not good enough at reading them to determine the max power draw that the connector can sustain. Also I don't even know if it is working, although it seems to be correctly connected to the CPU. I have passed your idea to our XL product manager and he said he will have a look (and I presume ask around if it is indeed possible).

toniklahan, 2025-04-12:
> Hi this is a cool idea (pun intended). I will try to talk with someone internally about it. The FAN2 connector has only 5V connected, but that shouldn't be a problem.

only 5 v ? are you sure ?

Jawaswag, 2025-04-28:
I support this, please. I'm in a tropical country where the ambient is around 30C. When I print ASA, I keep getting MCU overheat errors.