prusa3d/Prusa-Firmware-Buddy #4483: [ENHANCEMENT] Core One parking direction after print finished

Ondrashek, 2025-02-17:
### Printer model

Core One

### Describe the enhancement

Due to the Core One's official camera being positioned at the front left corner, it's better to modify park an print head at the rear instead of the front as default. This ensures the camera has a clear view of the print result when the job is finished.

CZDanol, 2025-02-17:
Hello, there are more practical reasons for parking the tool in the front:

1. It is when homing happens. Parking in the back would mean that we would have to move between front and back. Parking needs to happen at the front because of sheet detection.
2. It is more practical for maintenance reasons, filament changes (cleaning the ooze) and such

Ondrashek, 2025-02-17:
Your reason for parking the print head at the front is valid but should remain optional.

1. The printer performs homing at the start of a print, so the initial position of the print head doesn't matter.
2. For maintenance, users can manually move the print head or use a "Maintenance" command from the menu.

I still believe that having a camera view of the print when the job is done is more important than the other reasons.

CZDanol, 2025-02-21:
All right. Internal ticket: BFW-6820

danopernis, 2025-04-10:
@Ondrashek this should be fixed in 6.3.1 firmware

Tupson444, 2025-05-31:
>I would actually prefer it if it were at the front so that I could clean the nozzle more easily

Me too. But it would be better to actually allow users to configure what they want, by changing end gcode in the slicer. However, this is not doable because _after_ executing end gcode, the printer does a hardcoded end procedure (so whatever you put in the slicer, it won't matter). See #3785

CZDanol, 2025-06-24:
Since 6.3.1, we are now parking back, but only after print. Configurable parking positions are subject to another issue (#4319 / BFW-6394/BFW-6854), so I will be closing this issue.