prusa3d/Prusa-Firmware-Buddy #4485: [BUG]

Tworck, 2025-02-17:
### Printer model

MK4

### Firmware version

6.1.3+7898 MK4 [S][A]

### Upgrades and modifications

None

### Printing from...

Prusa Connect

### Describe the bug

Describe the bug
Blue Screen of Death (BSoD) occurred shortly after sending a print job via PrusaSlicer-2.8.0-beta1 through Prusa Connect. The Prusa Connect tab in the slicer didn't seem to fully load. Checking the printer showed the below error. It happens when I boot the printer via a remote power relay. the power relay is simply turned on/off via the wifi. Turning the relay on provides power to the printer.

BSOD text (using OCR, image also attached):

BSOD

stubs.cpp:5
aborted
task:defaultTask
SP:10005e90 LR:0802b91d PC:0802b91c
CPUID:04442803 ICSR_:00020200 VTOR_:fa050300
SCR__:00000210 SHCSR:00010000 DFSR_:e000ed34
AFSR_:00000030 ADR__:00100030
bot:10005e3c top:100051ac stack:
0802b91d fffffffc 20003d50 000000000 000000008
000000000 080300e3 ffffffffc 20003d50 000000000
000000008 000000000 080300e3 000000000 10005e5c
00041000 000000000 000000001 0802d701 000000000
000000001 000000000 00100402 000000001 0802fce3
6.1.3+7898 MK4 [S] [A]

[image]

### Files

will provide crash dump via reports email

CZDanol, 2025-02-18:
Hello, at this moment, we support the 6.2 and 6.3 versions. Please try updating first.

Tworck, 2025-02-18:
PR31333079426

CZDanol, 2025-02-18:
?

closed 2025-04-29 by github-actions[bot]