prusa3d/Prusa-Firmware-Buddy #4486: [BUG] Spontaneous reboot on filament change

mix579, 2025-02-18:
### Printer model

CORE One

### Firmware version

6.3.0+10073

### Upgrades and modifications

None

### Printing from...

USB Drive

### Describe the bug

Received the CORE One today. On the first filament swap, the printer rebooted after ejecting the filament. Another report of the same issue is here: https://forum.prusa3d.com/forum/prusa-core-one-hardware-firmware-and-software-help/coreone-spontaneous-reboot-on-filament-changes/#post-736972

### How to reproduce

Don't know yet. Will watch and report when it happens again.

### Expected behavior

No reboot

CZDanol, 2025-02-18:
Hello, do you have the door sensor enabled or disabled?

mix579, 2025-02-18:
Disabled

CZDanol, 2025-02-18:
Thank you, I've created a ticket BFW-6801 to collect the data

trackzero, 2025-03-24:
Same reboot on filament change issue, though I had the door sensor enabled.   I'd changed filament, it purged to the new color, and started moving back towards the model. I opened the door to grab the excess dangling from the nozzle, and it rebooted. Then it resumed in the wrong place. 3 hours into a 3:20 print. :/

closed 2025-06-02 by github-actions[bot]

Prusa-Support, 2025-06-06:
I'm reopening this issue in the hope that, if the problem persists or affects more users, we will have another chance to collect additional information.

Michele Moramarco
Prusa Research