prusa3d/Prusa-Firmware-Buddy #4487: Impossible Print? Repeated Print Failures - Arctos Robot X Pulley

Luke2791, 2025-02-18:
### Description of the bug

I have tried time and time again to print this part with no success. There are no apparent reasons (to me) why it should not be possible. This part has been successfully printed on many other printers (it’s part of a big open source project)… so I am surprised the MK4S has such a hard time with it. Others have printed this and gotten the same result (failed print due to layer shift)

My hope is that this is something that either firmware, software, or slicing can solve - and not a fundamental limitation of the printer’s hardware.  Please advise if you are able! 

**For details, please see this Prusa Forum post and the subsequent conversation:** 
https://forum.prusa3d.com/forum/english-forum-original-prusa-i3-mk4s-how-do-i-print-this/impossible-print-repeated-print-failures-arctos-robot-x-pulley/

Thanks, -Luke

[Prusa Forum Post - Impossible Print.zip](https://github.com/user-attachments/files/18741185/Prusa.Forum.Post.-.Impossible.Print.zip)

### Project file & How to reproduce

This "X Pulley" 3mf file will not print on my printer without a layer shift in the geared/toothed section of the part:

[Prusa Forum Post - Impossible Print.zip](https://github.com/user-attachments/files/18741235/Prusa.Forum.Post.-.Impossible.Print.zip)

### Checklist of files included above

- [x] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.9.0

### Operating system

macOS Sonoma Version 14.7.1

### Printer model

MK4S Kit (+ Enclosure Bundle)

Mirek1953, 2025-02-18:
Another finding: I managed to document that the "impact" sound I wrote about above has nothing to do with the nozzle hitting an already printed filament on the board. This sound, and we can say "bang", is made by the printer itself in FW 6.2.0. I simulated printing without a filament in the extruder, so the printer was running idle and this sound sounded exactly at the moment I described above without the nozzle hitting anything. When using FW 6.1.3, the described transition is absolutely smooth without any disturbing sounds. During tomorrow (19.2) I will post here the videos from this attempt. And I also made experimental prints with various FW:
FW 6.2.0                 failed
FW 6.2.0 RC1         failed
FW 6.2.0 alpha2     seems to be fine and there are no shifts
FW 6.1.3                perfectly fine.
I'll do experiments with FW 6.2.0 alpha1 tomorrow, but I think it's going to be fine.
With these attempts, I try to make it easier for developers to find a defect and narrow the space where the defect occurred. I hope they will appreciate it..  :-)  :-)

Luke2791, 2025-02-18:
@Mirek1953 you have gone above and beyond! This is fantastic. Thanks for your help in transferring the issue and following up on your experiment results. Appreciate your help with this too, @DanielBryars . 

@CZDanol as requested, I have attached a sample 3mf file that does not print correctly on my printer. Please note - the layer shifts are small - usually 1-2mm (not 10+mm as some have experienced). However, they still cause the print to be useless since these small layer shifts mess up the gear teeth. Thanks for your help. 

[X pulley Failed Print.3mf.zip](https://github.com/user-attachments/files/18853961/X.pulley.Failed.Print.3mf.zip)

Mirek1953, 2025-02-19:
Attached are the promised videos of attempts to run the printer without filament. And I managed to install the latest FW 6.2.1 and do a test print. It turned out exactly as expected: The print failed and the layer shifted. 

[pokusFW.zip](https://github.com/user-attachments/files/18863675/pokusFW.zip)

Luke2791, 2025-02-21:
Wonderful to hear, @danopernis ! Please keep us posted as/when you are able. Look forward to seeing the fix!

Also - for reference - I recently reverted my firmware to 6.1.3 - it is was able to print the aforementioned print just fine. So I can now confirm what the others have suggested.

Luke2791, 2025-03-05:
@danopernis this is great news! Thanks for the update. Have you (or anyone else) been able to get a successful print of the above 3mf file with 6.2.2 firmware? Just wanted to confirm that it works with this “specific” file as well. Thanks again!

CZDanol, 2025-03-05:
We were testing a bunch of specific prints people reported as problematic, not sure if this was one of them.

Luke2791, 2025-03-09:
hi @CZDanol and @danopernis . The file attached by me above, [X pulley Failed Print.3mf.zip](https://github.com/user-attachments/files/18853961/X.pulley.Failed.Print.3mf.zip), is the original print file that this issue was opened with. It would be a shame for this thread / issue to be closed without that ever being tested and confirmed fixed.

@CZDanol I noticed you said, "...not sure if this was one of them." So, that is why I am responding. Could someone there are Prusa **_confirm_** please that this 3mf now prints fine on the 6.2.2 firmware??? If so, I will happily close this issue/thread :). 

Again - not being able to print this specific part (and others like it) was the reason I opened this issue to begin with. Unfortunately, I do not have any more extra filament left to burn/waste on troubleshooting this part (it's cost me quite a lot already). Thanks for your help with this!

CZDanol, 2025-03-10:
All right, I will run the print for you :)

Luke2791, 2025-03-10:
Thanks for seeing this thru to the end, @CZDanol ! Appreciate it very much!

CZDanol, 2025-03-11:
Okay, so I printed the 3mf you provided (X pulley Failed Print.3mf) on MK4S 6.2.2 with PETG and I didn't get a layer shift.

I have however noticed that you're using gyroid infill and there currently is a known bug where gyroid infill can cause a numeric error accumulation, resulting in a layer shift, so I would perhaps advise you to use a different (for example cubic) infill.

[image]
[image]
[image]

mix579, 2025-03-11:
Now I'm very confused... I thought this fix was to address this very issue (also in #4465)? Are you saying it HASN'T been fixed? Then what is this fix that was pushed out in 6.2.2?

CZDanol, 2025-03-11:
The fix was supposed to fix layer shifts regression introduced in 6.2.0-RC1. So whatever printed fine in 6.1 and did not print fine in 6.2.

The gyroid infill is a separate issue that we have discovered in 6.2 (and thus mentioned in the release notes), but it has been in the code since the beginning. Fixing that requires some changes that we are not able to do as a part of 6.2.

Many people jumped to a conclusion that these two things are the same issue, but they are not.

Luke2791, 2025-03-14:
@CZDanol if I understand correctly
1.  The aforementioned 3mf file is printing fine now (even with gyroid infill) 
2. The issue with gyroid infill only *sometimes* causes issues - and is separate/unrelated. 
3. The latest firmware has resolved the issue this thread was opened with and can closed. 

If all of this is correct - feel free to thumbs up or respond to this message and I can close the thread.
Thank you all!

NickSica, 2025-04-03:
I've tried to print gears similar to this in the past week with gyroid infill and get a massive layer shift around the same layer. Could this be part of this issue or the gyroid issue?