prusa3d/Prusa-Firmware-Buddy #4488: [BUG] BSOD event_groups.c:531 when turning MK4s on with MMU3 attached

DavidGeorge528, 2025-02-19:
### Printer model

MK4s

### Firmware version

6.2.0, 6.2.1

### Upgrades and modifications

MMU3

### Printing from...

N/A

### Describe the bug

Blue Screen of Death (BSOD) when turning the printer on with MMU3 attached. Disconnecting the MMU3 allows the printer to boot normally. I've followed [this BSOD article](https://help.prusa3d.com/article/blue-screen-of-death-bsod_689421), and have factory reset the printer multiple times.

[image]

The printer _has_ booted with the MMU attached, but I have not found a repeatable way to achieve this. The first time was after a factory reset when I was setting the printer back up, however on the next boot the BSOD came back. I cannot see any visible damage to the cable, the Buddy board, the MMU board or the power board on the MMU.

### How to reproduce

Turn on printer with MMU3 attached

### Expected behavior

Printer turns on as normal

### Files

PR31256206018

DavidGeorge528, 2025-02-28:
The support ticket I raised concluded that it was a bad main board for the MK4s and a replacement was sent which seems to have fixed the issue. 

Though there might still be information to be learned from the dump

danopernis, 2025-03-05:
Good to hear that your problem was solved.

From analyzing the dump, it looks like the memory was corrupted even before the program got to the place where it crashed. The dump therefore unfortunately didn't reveal the root cause of the problem, but thank you for sending it anyway.

The memory corruption issues are luckily quite rare. They could be partially mitigated by using ARM's memory protection unit, but it is quite difficult to retrofit into the codebase. We will try to get there eventually.

closed 2025-03-05 by danopernis