prusa3d/Prusa-Firmware-Buddy #4489: [BUG] Turning on phase stepping sometimes causes crash detection

sp847242, 2025-02-19:
### Printer model

XL Multi-tool

### Firmware version

6.2.0-8816

### Printing from...

USB drive

### Describe the bug

This occurs with both my own printer and one at work, both running 6.2.0 firmware, though it happened in prior firmware versions as well. Both are dual-head versions.

Occasionally, when turning Phase Stepping off during a print, when the printer pauses to do the mode change, it will make a "clunk" sound and report that a crash has occurred. It then rehomes itself and resumes printing, with phase stepping now off. 

### How to reproduce

All it takes to cause it is to turn Phase Stepping off in the printer's menu, _however_, I haven't been able to get it to happen reliably. Most of the time it does the switchover just fine. My best guess is that it has maybe a 5% chance of happening. I haven't been able to see a pattern in terms of when it occurs, such as if it's on a perimeter, or a straight line, or a curve, etc.

(The reason I switch Phase Stepping off is to get crash detection back when I may want it, either for tricky prints, or if I can't be near the printer. If I'm going to be near the printer, or if it's generally a simple print that's not prone to issues, then I'll have Phase Stepping on. So I may be toggling it more often than most.)

CZDanol, 2025-02-19:
Hello, thank you for your report. Internal ticket: BFW-6809

sp847242, 2025-04-25:
> Hello, thank you for your report. Internal ticket: BFW-6809

Re: "may be considered 'stale'" notice from github-actions: 
If this has this internal ticket #, do I still need to keep replying every so often to keep it from going stale?

CZDanol, 2025-04-25:
Assigned myself to keep the stale bot away

closed 2026-04-13 by github-actions[bot]