prusa3d/Prusa-Firmware-Buddy #4490: Added option to calibrate HotEnd PID. from the printer's menu.

Mirek1953, 2025-02-19:
### Printer model

MK4, XL, CORE One, MK3.9

### Describe the feature

Since new and new versions of heating blocks are appearing (whether they are copper blocks: https://levendigs.com/collections/prusa-mk4-parts-upgrades/products/nextruder-heaterblock-plated-copper-for-prusa-mk4-xl or light ceramic heating cartridges: https://trianglelab.net/products/chc%C2%AE-for-prusa-mk4-xl?VariantsId=11647) and these blocks have different heating characteristics, it would therefore be advisable to introduce the possibility of calibrating their PID characteristics. The possibility here is using M301 but unfortunately Prusa has disabled the M500 code for storing the characteristics in EPROM for some reason with MK4.  So it would probably be best to add a new option called PID Calibration to the Nextruder section of the menu. The control could be in the same style as the InputShaper calibration, i.e. display the setpoints first and use the Calibrate PID option to start the calibration, preferably using a wizard with a warning that the extruder will heat up. When the calibration process is complete, display the new P,I,D values and ask if the old values should be overwritten and saved similar to the InrutShaper calibration.

### Expected functionality

Improved user experience when using Prusa printers.

LarGriff, 2025-03-29:
Ditto

rinkek, 2025-03-31:
Pretty please!

rinkek, 2025-06-02:
Would be nice.

m-s-y, 2025-08-06:
I’d like it too

3d-gussner, 2025-08-06:
Created an internal ticket.

As a workaround you can
- run the PID tuning for your hotend via USB cable and serial terminal like Putty
  - copy the new PID values
- Add to your PrusaSlicer profiles in the Start-gcode section the `M301 P<your new value> I<your new value> D<your new value> to set it for the time the printer is running.

Tupson444, 2025-08-06:
>for the time the printer is running

Unrelated to this issue, but why Buddy firmware doesn't have an option to save values, like Marlin? Many things should be changed once, and not depend on the settings in the file, this being a good example (PID has nothing to do with the gcode file, but with hardware).

aknezevi-work, 2026-02-04:
This would make printer upgrades and factory resets so much easier. It would also bypass the need to connect to the printer with a cable in an awkward way.

redfox-mx, 2026-09-01:
.

Tupson444, 2026-11-02:
not stale