prusa3d/Prusa-Firmware-Buddy #4492: [BUG] MMU Overcurrent kept triggering whenever powering back on right after powering down

5jvm0u4, 2025-02-20:
### Printer model

MK4S(Stock&New)+MMU3(upgraded from MMU2S)

### Firmware version

6.2.1+8833 & 3.0.3

### Printing from...

Nothing

### Describe the bug

I consider it a bug because if I power down the whole system for 5 minutes and power it back on there's no overcurrent problems, and also, the beeping is really annoying, please limit it to five beeps or under, it's enough, I know there's a problem when my whole screen turned red.

### How to reproduce

Everything when I power back on right after powering down, through the power button on the power supply, it happens, and everytime I wait for 5 minutes, it boots without any problem

bigman212-help, 2025-02-24:
Yep, I'm on my 4th buddy board and 3rd mmu board and pd board. Not sure how I'm going to get a refund on my mk3.5s since it's hard to return the items without destroying the printer and mmu2s doesn't work on the mk3.5s. Shame on prusa for selling a faulty product. Will probably go with a charge back on Amex and switch to Bambu.

bigman212-help, 2025-02-24:
Sometimes it's 5 minutes, I've gotten one where even after 24 hours it wouldn't boot and I had to take out the printer from my lack enclosure and unplug the whole mmu

LarGriff, 2025-03-03:
This seems to be the same issue…
https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3947
A solution has been found by a customer, but Prusa hasn’t yet acknowledged it.  😕

5jvm0u4, 2025-03-03:
> This seems to be the same issue… [#3947](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3947) Caution: Very long read with lots of opinions. Skip to the end and work backwards. A solution has been found by a customer, but Prusa hasn’t yet acknowledged it. 😕

It seems to be a different problem, I got red screen of death, not blue. Like this one
https://www.reddit.com/r/prusa3d/comments/1ig2lc3/help_mmu_over_current_error/?rdt=51263

5jvm0u4, 2025-03-03:
Your right, I'll leave a commant there in case they eventually fix the problem and want to close duplications

closed 2025-03-03 by 5jvm0u4