prusa3d/Prusa-Firmware-Buddy #4493: [BUG] When any error triggers related to MMU, there's no abort option, only retry

5jvm0u4, 2025-02-20:
### Printer model

MK4S & MMU3

### Firmware version

6.2.1+8833 & 3.0.3

### Upgrades and modifications

None

### Printing from...

nothing

### Describe the bug

As mentioned, when an error triggers, ex: fsensor too early, I could only retry, not abort and do something else to fix the problem, and I don't like resetting the printer because it'll sometime cause MMU overcurrent to appear and I had to power off my printer for 5 minutes doing nothing.

DRracer, 2025-03-03:
While the error is being displayed, the printer stopped its operation in order to allow for manual fixing of the issue. Adding another button (like the "Abort" you proposed) would only be good for stopping the whole print. However, having another button on the screen would also bring potential danger of pressing such a button accidentally and ruining the print unintentionally. Therefore we prefer having just a single "Retry" button which instructs the printer, that you have manually resolved all the issues which prevented normal operation and the print can now continue.

5jvm0u4, 2025-03-03:
Yes, my intention is to abort the whole print, because my printer is in an enclosure, there's no way I could fix anything while keeping it plugged in.

5jvm0u4, 2025-03-29:
If prusa are planning to fix this, please consider fixing this as well #4545

LarGriff, 2025-03-29:
👍🏻

5jvm0u4, 2025-05-29:
Seriously?

5jvm0u4, 2025-09-03:
Just uninstall my mmu, too much of a headache, not worth the time and effort, if I really want a multi-color print I'll probably just buy a A1 combo and use it as a disposable.
An A1 mini standalone cost the same as the shipping for a Core One here at my place, that's insane.

tensau, 2025-09-20:
Nothing new on this feature request?

5jvm0u4, 2025-09-22:
I feel like if Prusa does not intend to do it this way, at least mark this bug as "work as intended" or something else so that we know it's never going to be fixed, rather then waiting for eternity.

5jvm0u4, 2025-11-25:
Here we go again