prusa3d/Prusa-Firmware-Buddy #4494: [BUG] Upgrading from 6.1.4 to 6.2.1(OTA) wipe out filament sensor calibration infos

5jvm0u4, 2025-02-20:
### Printer model

MK4S + MMU3

### Firmware version

6.2.1 & 3.0.3

### Upgrades and modifications

none

### Printing from...

PrusaLink

### Describe the bug

I upgraded through Prusa Connect, and while it's upgrading, a MMU overcurrent triggered(as usual), I powered it off for several minutes and power it back on, it showed flashing ESP, and started without any problem.
I noticed my hardware info "HF hotend" is set to off by default, which I learned from the release page is fixed from 6.2.0 to 6.2.1, I switched it to on and didn't think too much, but when I wanted to print, "fsensor too early" kept triggering and after some troubleshooting I found out it's the filament sensor calibration infos being wipe out and I need to redo it.

CZDanol, 2025-02-28:
Hello, were filament sensor calibrations the only thing that got wiped out? If so, that would rather indicate that you changed clicked on the "Extruder type" option in the hardware/printer setup menu. Changing between standard and MMU rework clears the fsensor calibrations (intentionally and correctly).

5jvm0u4, 2025-02-28:
I didn't change that, but as you mentioned this, I did miss-clicked the printer type option that says [MK4S] to [MK3.9], is that what it was? If so, I would suggest adding a warning message there to at least notify user what will be wiped out if proceeding, so people won't accidentally click on it when not intended.

CZDanol, 2025-03-11:
MK4S -> MK3.9 should not invalidate the fsensor.

closed 2025-05-19 by github-actions[bot]