prusa3d/Prusa-Firmware-Buddy #4495: [FEATURE REQUEST] Prusa XL support Hackerboard

williamrehndell, 2025-02-21:
### Printer model

Prusa XL

### Describe the feature

Prusa printers from the MK3.5 up to the Core One currently support the hackerboard, but unfortunately not your largest printer—the Prusa XL. 

It would be perfect to have support for the Prusa XL so that you can integrate custom extra features, as you describe so impressively in the GPIO "Hackerboard" product description. 

Some of these features include timelapse and an LED that can indicate whether tool head 1 or 2 is in use, among others.

### Expected functionality

Prusa XL support for GPIO "Hackerboard"

CZDanol, 2025-02-27:
Internal ticket: BFW-6850

CZDanol, 2025-02-27:
Hello, looking at the scheme of the XL, it does not seem that there is an I2C wired to any connector except for the LCD. We will give it some more thought, but it might not be possible. Ideas welcome.

CZDanol, 2025-02-27:
Hello, looking at the XL schemes, I don't see any connector that has I2C wired up, except for the LCD one. Ideas welcome, but this might be a stopper.

williamrehndell, 2025-02-27:
Maybe an adapter could be used to connect both the screen and the hacker board?

CZDanol, 2025-02-28:
I don't think that would work, both economically and practically. It would mean:

1. Designing and manufacturing another splitter board that would have two LCD connectors.
2. Manufacturing another short LCD cable that would bridge the new splitter board and main board.
3. Extending already long cable to the xLCD, and adding two more connectors on the way, making it more prone to comm errors. Let me remind you that USB goes through that cable.

As for the XL enclosure comment by @Graloth - the board on the XL enclosure is actually just a routing board for the fan, led and USB ports.

williamrehndell, 2025-02-28:
Okay, it’s unfortunate if this might not be possible to resolve. Considering that a printer described as “The Original Prusa XL is the most innovative 3D printer on the market”, one would expect it to at least be capable of the same operations as the MK3.5—especially since the hacker board is a great solution for controlling additional functions that many users have requested. 

I hope you find a workaround :)

Graloth, 2025-02-28:
@CZDanol Would it be possible to use the USB-C port (either on the board with the wifi chip, or on the enclosure board), and route the needed I2C communication through that, and have an "XL hackerboard" just connect to the XL that way?

I know that's not really ideal, since it could cause problem if you directly repurpose pins for something that breaks normal USB-C functionality.

Alternatively you could create a USB device class that tunnels the I2C communication without breaking the USB spec, this could then also implement some sort of handshake so the printer knows it should switch to I2C communication when that device is plugged in.
And that device could then have the connector needed for using the hackerboard to avoid having to maintain difference versions of it for different printers.

CZDanol, 2025-03-01:
For using the USB-C pins - I am pretty sure that although the connector is USB-C, we only have USB 2.0 connected, which means 5V, GND and D+/D-. And even that is maybe not wired directly to the MCU, but goes through a dedicated USB chip.

If we get into realm of creating an "USB Device class", that means that we need to have a microcontroller on the "hackerboard" and at that point, there is no point in using I2C. You might as well have the extension control the printer over the serial line the printer has on that port through gcodes.

Graloth, 2025-03-10:
@CZDanol Ah, was hoping it had a more direct connection with the MCU 😅 

I think the most reasonable solution would then be for people to use an ESP32-S3 (or similar that support running as a USB host device to read serial output from the printer) that can be powered over the same USB port it retrieves data over. Since then you'd just have to code in your own triggers based on what the serial output provides.
Should very much be possible with some tinkering.

Might give it a go, but can't make any promises, though if I get some basic functionality working I will share it so those with more knowledge can pick up from there 😁

BlackRasberry, 2025-03-28:
Is it unreasonable to rev the board? I have some significant functionality I would like to add using a hackerboard. I would pay for a new board that addid this functionality. Isn't that part of the reason the board is 3 parts with pcie connectors? Easy revisions on part of the board?

CZDanol, 2025-03-29:
Making a board revision just to add the hackerboard support wouldn't be economical for us. However when there would be a time for a new board revision, I bring this up.

As for the unpopulated connector, that can definitely be used for something - it has a RS232/modbus on it, but not for the GPIO expander, because it doesn't have I2C.

Currently, possibly the best way to expand XL IO would be to get something like Raspberry, connect to the printers serial over USB and read outputs/send commands through that.

williamrehndell, 2025-03-29:
Alright, thank you so much for taking a closer look at this! And thanks to everyone else who shared their suggestions as well! We’re hoping to see a Prusa XL(s) soon :)

closed 2025-03-29 by williamrehndell