prusa3d/Prusa-Firmware-Buddy #4496: [BUG]

VSCustoms, 2025-02-23:
### Printer model

MK4S

### Firmware version

6.2.1,6.3.0

### Upgrades and modifications

None

### Printing from...

USB Drive

### Describe the bug

I seen this issue reported for Mini+ but my MK4S is a nightmare recently and maybe it’s an issue in slicer but I have tried different copy’s there too. 

Random layer shift when using fuzzy skin. 

My setting on fuzzy skin and .1 thickness and .2 distance. I have tried 6.2.0 RC1, 6.2.1 , 6.3.0, formatting my USB, Slicer 2.9.0, 2.9.1 Alpha1, and it does not matter. 

 Filament  stuck sensor is off because that just does not work. 
 Nozzle is not clogged and has been changed to try, I have recalibrated it 3 different times. 

 The only time I got a perfect print was no fuzzy skin. 2.9.0, 6.2.1 firmware

[image]

### How to reproduce

Just print with fuzzy skin and it replicates every time. Print with it off and it’s perfect every time.

murk-sy, 2025-02-23:
It is the same issue so you can close this as duplicate, I just originally reported it as for the Mini since that's where I observed it.
- https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4465

Workaround is downgrading to 6.1.x or avoiding any "wobbly" extrusions"
Supposedly a hotfix should be released within a few days

VSCustoms, 2025-02-23:
> It is the same issue so you can close this as duplicate, I just originally reported it as for the Mini since that's where I observed it.
> 
> * [[BFW-6795] [BUG] Layer shifts on 6.2.0 #4465](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4465)
> 
> Workaround is downgrading to 6.1.x or avoiding any "wobbly" extrusions" Supposedly a hotfix should be released within a few days

Ok. Just wanted to make sure they know the same issue is showing up on the MK4S also. 
 I can’t downgrade 6.1.x as it causes BSOD on MK4S with MMu3. So I am kind of stuck.