prusa3d/Prusa-Firmware-Buddy #4497: [BUG] USB Error

Gik-0101, 2025-02-23:
### Printer model

XL Multi Tool

### Firmware version

6.2.1

### Upgrades and modifications

Fan Shroud 360, Enclosure

### Printing from...

USB Drive

### Describe the bug

I had a USB failure twice during a 2 day 8 hour print job. Unplugging and plugging it back in saved the print

, but of course the mistake happened during the night. 
So please excuse the bad photo. 

I would say that the USB error in the current FW is still not 100% fixed.

[image]

CloudStrife1983, 2025-02-25:
@Gik-0101 It looks very much like your USB stick is simply broken. The Prusa sticks often break.

CZDanol, 2025-02-25:
We might have a relevant internal ticket BFW-6594

Gik-0101, 2025-03-01:
Exhausting.....
I'm currently failing at a multi-tool print of the "Earth Lamp". This is supposed to take 1d 8h. Unfortunately, the printer always freezes between 1-3% of the print progress. The progress continues normally in connect and on the LCD, but the printer stuck.
It always freezes when a tool change should be done.
Pausing the print and then continuing it doesn't work. Turning the printer off and on helps for a few layers.

I've tried:
FW 6.2.0
FW 6.2.1
Bgcode
Gcode
3 different USB sticks (Intenso, Sandisk, Orig. Prusa)

Of course directly from the Prusa Slicer to the USB and ejected it properly. Formatting is also done properly.

I don't know if this is related to the error, as I'm not getting any error message from the printer.

CZDanol, 2025-03-01:
That sounds like a different issue that you were describing in the beginning, am I correct?

Gik-0101, 2025-03-02:
Sorry,
I don`t know if this is a clearly different error. Maybe a combined error? 

I found an older bug report here on github and joined it:

https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3983
I also linked there a video, .3mf,....

Gik-0101, 2025-03-06:
Well... what can I say.

Replaced hardware with Prusa Support.
The non-functioning print now ran through to the end without any problems.

I think we can close the bug because it was triggered by defective hardware.

closed 2025-03-06 by Gik-0101