prusa3d/Prusa-Firmware-Buddy #4498: [BUG] m220 shows no effect

klapptnicht, 2025-02-24:
### Printer model

XL 5 tool

### Firmware version

6.2.0

### Printing from...

prusa connect

### Describe the bug

I have added a m200 Sn command after layer change: 

{if layer_z > 0.2}m220 S{180+int(layer_z/2.0)*10}{endif}

While the slices shows it as expected, the printer does not show any effect, nor does the display show anything else then 100%

Also generated gcode seems to be correct.

### How to reproduce

Using the model and following the instruction of: https://www.printables.com/model/342075-extrusion-test-structure, adding m220 as described above to the "after layer change gcode".

Alternatively for testing you may want to start with "m220 S50" and then after a couple of lines switch to e.g. 200%.

### Expected behavior

Changing the speed according to the command and also update UI and prusa connect status.

u89djt, 2025-02-24:
You need to zip the 3mf file to post it. A screenshot can save the devs time searching for it.

danopernis, 2025-02-24:
G-Code parser appears to be case-sensitive, can you try with `M220` instead of `m220`?

klapptnicht, 2025-02-24:
Ja, solved it. Apologies - after neither the slicer nor the gcode viewer complained I didn't even expect this being an issue.

closed 2025-02-24 by klapptnicht