prusa3d/Prusa-Firmware-Buddy #4499: [BUG] Prusa Connect Events Log Spam

muesli, 2025-02-24:
### Printer model

MK3.5

### Firmware version

6.2.1

### Upgrades and modifications

None

### Printing from...

Prusa Connect

### Describe the bug

I've created a custom material on my MK3.5. Ever since the Events Log is getting spammed with material change events, even though the machine is just sitting there being idle:

[image]

This happens roughly once every second(!).

CZDanol, 2025-02-25:
Hello, I might sound a bit silly right now, but where is this event log? 🤨

Gik-0101, 2025-02-25:
I can confirm that.

At least with the self-created filament presets in the printer.

@CZDanol 
It will appear at the bottom of the Prusa Connect dashboard in the event log.
It is also permanently displayed and updated under telemetry.
I haven't been able to test whether this also occurs with the standard filament settings in the printer.

[image]
[image]

CZDanol, 2025-02-25:
Okay, thank you guys for the info, we will look into it.

CZDanol, 2025-02-26:
Okay, I have talked this over with the Connect guys and they confirmed that this is a problem in the Connect, not in the firmware. They will let us know when it's fixed :)