prusa3d/Prusa-Firmware-Buddy #4501: [BUG] MK4S firmware 6.2.1 USB error

mt-mrx, 2025-02-25:
### Printer model

MK4S

### Firmware version

6.2.1+8833

### Upgrades and modifications

Quick Connect display cable https://www.printables.com/model/612910-mk4-lcd-quick-release-cable

### Printing from...

USB drive, Prusa Connect

### Describe the bug

Multiple crashes while printing via USB drive(s).

Just upgraded the printer to 6.2.1+8833 and started the first print.

**First try:**
- Using PrusaSclicer 2.9.0 to generate the gcode
- I uploaded the attached gcode file via Connect to the printer. Those are MMU3 parts.
- Printer is using the original Prusa USB drive
- The MK4S started printing without issue.
- At around 4-6% progress (before the infill layers begin) I glance a "USB file error" or something like that on the display and when I reached the printer it was already showing a bluescreen.
- After a Reset I got another bluescreen

**Second try:**
- Took the original Prusa USB drive and used a Linux system to recreated the partition table as msdos and **reformatted** a new partition with fat32.
- Rescliced the project.
- Transfered the gcode file to the usb drive.
- Verified the sha256 checksum of the file on my disk and on the drive, were identical.
- Moved the USB drive to the printer and it crashed again. This time I captured the bluescreen
[image]
- After a Reset I got another bluescreen, captured it.
[image]
- Another reset later I also captured a dump file which I will send to support afterwards.

**Third try:**
- Took a **different USB drive**, cleaned it, created a new filesystem, transfered the gcode file, verified checksum.
- This time I printed without filament to not waste my Prusament PC-CF
- Started printing same error/crash. Somewhere around 3%-6%.

**Fourth try:**
- **Downgrade to firmware 6.2.0** .
- Used the exact USB drive, without reformating it, as in the second try. Checksum of the gcode file was still correct.
- Printer printed the file in around 11hrs without any issues.
- Since then I printed more parts for the MMU3 and had no further issues.

I first thought that the quick connect cable I custom crimped for the display may have an issue suddenly, even though I'm using it since a couple of months and dozens/hundreds of print hours with it.
But since everything is working fine after the firmware downgrade my confidence in my crimped cable is pretty high.

I have not tried to upgrade to 6.2.1 yet again and retest (I want to get the parts for the MMU3 printed ;-) )
 

### How to reproduce

Just try printing the following file using firmware 6.2.1
[MMU3_gcode_file.zip](https://github.com/user-attachments/files/18970136/MMU3_gcode_file.zip)

gcode file checksum SHA256:
ce045cda9281de79e6532c19e00c286aa34a6d8cda06127dd953f223c2fcd80b

### Expected behavior

No crash while printing, obviously.

mt-mrx, 2025-02-25:
Ticket ID with the dump file: PR31404073922

mt-mrx, 2025-02-25:
Thanks for the quick feedback.

CZDanol, 2025-02-26:
Damn my tunnel vission, sorry. I only saw the connectTask stack overflow - which should indeed be fixed in 6.2.2. The Watchdog Reset thingy we do not know about. We will investigate the dump you sent - is it the one from the Watchdog Reset? If not, it would be very handy to get one from that screen.

mt-mrx, 2025-02-26:
The watchdog did not produce a dump. (I think)
- The system crashed while printing with the USBH_MSC... screen, I reset it
- after it booted it was only briefly showed the menu, if at all
- no interaction from my side, did not offer to save a crash dump and crashed again with the "connectTask" screen.
- Another reset later it offered me to save the crash dump, that's the file I sent you. Not sure from which crash the dump is.

CZDanol, 2025-02-26:
Oh, right, I see. Hmm. I will look on the crash dump when I get some time.

mt-mrx, 2025-02-26:
I just tried to recreate the issue by upgrading again to 6.2.1 and printing the same gcode file I attached above with which I could crash the printer multiple times before. After 1:24 hrs printing without filament and reaching 11% I canceled the print. So far no issue.
I'll print something else and will stay on 6.2.1 to see if the issue may reappear.

mt-mrx, 2025-03-01:
Hi, small update. After upgrading to 6.2.1 again I printed 6 different jobs in the last 3 days with ~20hrs print time, so far everything looks good, not a single issue.

mt-mrx, 2025-03-04:
Hi, today I had another failed print with firmware 6.2.1 . I was not watching the printer while it failed so I cannot confirm if there was first a USBH_MSC... bluescreen, but here is what I observed:

- Printer was running firmware 6.2.1 and a ~5hrs print.
- When I checked on my printer in the morning after the overnight print it was stuck at showing the bootloader screen.
- The bootloader progress bar was all the way to the left.
- The printer had cooled down to ambient temperature so it was sitting at the boot loader screen for at least an hour.
- I hit the reset button.
- Printer restarted and I got immediately the following bluescreen after the bootloader finished loading.
[image]
- I hit the reset button again.
- Printer booted fine and offered me to save the crash dump, which I have.
- I can send the crash dump to support if requested.
- I downgraded again to 6.2.0

danopernis, 2025-03-05:
Thanks for the report. This should be fixed in newly released version 6.2.2

mt-mrx, 2025-03-16:
Hi, I upgraded to 6.2.2 a couple of days after it was released, since then I printed almost 1.5kg filament and a lot of hours.
Today I got a bluescreen again of the original type I reported further above, the USBH_MSC_Worker error.
Here is what happened.

- Printed a small, only 6g ASA, model directly from the USB drive. The gcode (not bgcode) was transferred via PrusaLink to the printer.
- The firmware stopped printing and complained about something like "gcode commands are being trasmitted too slowly".
- Extruder was moved to park position automatically.
- ~5s later the extruder was moved back to the model and continued printing
- another ~5s later the printer said that I "should plug in the USB drive again"
- A couple of seconds later the printer crashed and restarted without any message.
- Immediately after the bootloader completed loading I got the following USBH_MSC_Worker error displayed.

[image]

- I hit the reset button and after the restart I was offered to save the crash dump, I confirmed doing so, but the printer crashed immediately again, with another USBH_MSC_Worker error.

[image]

- Resetting the printer resulted again in the same error. I did an AC power cycle and checked the contents of the USB drive for the dump file, but apparently it wasn't written.
- I reinserted the USB drive and printed the identical file again (didn't overwrite or reuploaded it) without any errors.

So it looks like the ConnectTask error was fixed but the my issue I reported further above apparently was not, I still can get USBH_MSC_Worker bluescreens.

CZDanol, 2025-03-16:
Hello, yes, that makes sense. Now we just need to get the "real" crash dump :)

The dump might still be saved in your printer. Can you try doing "Settings → System → Save Crash Dump"?

mt-mrx, 2025-03-16:
Hi, I managed to do save the crash dump. But it wasn't easy. I was already printing the next job when your message came in.
- So after the job finished I just went ahead and did "Settings → System → Save Crash Dump" ...
- The printer crashed immediately again with the USBH_MSC_Worker bluescreens, no dump was written.
- During power on it offered me again to write the crash dump, I said yes and the printer crashed again.
- Next I prepared a new USB drive (8GB/USB3) with fat32 filesystem.
- Onto that drive the crash dump could be saved.
- I took the originial Prusa USB drive and did a filesystem check, fixed some errors, see below.
- After the filesystem errors were fixed I could use the originial Prusa USB drive to also save the crash dump without getting a printer bluescreen.
- I'm sending you two crash dumps via email.

```
root@libra:~ # fsck /dev/sda1
fsck from util-linux 2.40.4
fsck.fat 4.2 (2021-01-31)
FATs differ - using first FAT.
/IntegratedRewinder_P2_0.4n_0.2mm_ASA_MK4S_1h33m.gcode
  Contains a free cluster (16022). Assuming EOF.
/IntegratedRewinder_P2_0.4n_0.2mm_ASA_MK4S_1h33m.gcode
  File size is 10135296 bytes, cluster chain length is 2719744 bytes.
  Truncating file to 2719744 bytes.
/IntegratedRewinder_P3_0.4n_0.2mm_ASA_MK4S_3h44m.gcode
  File size is 29757945 bytes, cluster chain length is 16384 bytes.
  Truncating file to 16384 bytes.
Reclaimed 5285 unused clusters (86589440 bytes).
Free cluster summary wrong (1864431 vs. really 1869716)
1) Correct
2) Don't correct
[12?q]? 1

*** Filesystem was changed ***
The changes have not yet been written, you can still choose to leave the
filesystem unmodified:
1) Write changes
2) Leave filesystem unchanged
[12?q]? 1
/dev/sda1: 44 files, 49281/1918997 clusters
root@libra:~ # fsck /dev/sda1
fsck from util-linux 2.40.4
fsck.fat 4.2 (2021-01-31)
/dev/sda1: 44 files, 49281/1918997 clusters
root@libra:~ # sync
```

mt-mrx, 2025-03-16:
Hi Daniel,

after some investigation it's likely that the crashes were caused by me. When installing my other USB drive I noticed that its activity LED was flickering all the time even if the printer wasn't doing anything. Also the crashes kept getting worse but I realized when the printer was stuck during the boot loader I could get it to continue to boot by removing the USB drive.
I also got multiple filesystem corruptions.

I removed the printer from the enclosure and rechecked my modification, which I mentioned above, the Quick Connect display cable from [here](https://www.printables.com/model/612910-mk4-lcd-quick-release-cable).

I measured all pins of the cable again and some of them had a higher resistance than the original cable, around 7-15 Ohm. The resistance may have increased over time because of vibrations and/or the changing temperatures. 

Anyway, I replaced that quick connect back with an original direct cable where I measured each pin at around only 0.16-0.22 Ohm.

So hopefully that fixes this and if this issue was really caused by the modification I'm very sorry for wasting your time.

CZDanol, 2025-03-18:
Hello, no problem :) Glad that your issue got resolved.

The BSOD still shouldn't ideally happen, so I think it's possibly still worth looking into it for us. What e-mail did you send it to?

mt-mrx, 2025-03-18:
I replied to the email I got as response when I sent my initial crash dump. That reply went to info@prusa3d.com should I send again to reports@prusa3d.com ?

CZDanol, 2025-03-18:
Internal ticket to investigate the dumps: BFW-6939