prusa3d/Prusa-Firmware-Buddy #4502: [BUG] Spontaneous reboot when opening door early as a print is lowering on the z-axis after a completed print

devzero, 2025-02-27:
### Printer model

Core-One

### Firmware version

6.3.0+10073

### Upgrades and modifications

None

### Printing from...

WiFi

### Describe the bug

I've found that consistently, when I don't wait until a completed print moves all the way down the Z-axis once it is complete but instead hastily open the door, the embedded computer reboots instantly every time I do this.  It's not the worst behavior in the world, but it doesn't seem intended and I hope that it's not a wiring/electrical issue.

CZDanol, 2025-02-27:
Hello, thank you for your report. We know about this and it should hopefully be fixed in 6.3.1.

It's not a wiring/electrical issue. When the door is open, the printer shouldn't do any Z moves more than a few mm to prevent squashing the hand. What you see is a last-resort stop, because the firmware didn't manage to block the movement through other means.

Patronics, 2025-11-13:
This also seemed to happen in 6.3.4 when I opened the door at the start of a print (to remove some ooze from the nozzle)