prusa3d/Prusa-Firmware-Buddy #4505: [BUG] Z height too low when resuming after a pause to insert magnets

tylerdola, 2025-03-01:
### Printer model

CORE One

### Firmware version

6.3.0+10073

### Upgrades and modifications

None

### Printing from...

Prusa Connect

### Describe the bug

**Setup**

When attempting to print nameplates with embedded magnets using PLA, I insert a pause in Prusa Slicer 2.9.0 before layer 12. The CORE One pauses, homes the extruder, and significantly lowers the print bed, waiting for my actions. It also appears to turn off the nozzle heat. After inserting the magnets and clicking resume, the print bed rises until the nozzle touches the part and continues printing layer 12.

**Problem**

The subsequent layers are too low and of poor quality. The print head strikes the layers below, and it takes several layers to even partially smooth out. The print never fully recovers, as subsequent layers remain lower than they should be.

**Pictures**

[image]

[image]

### How to reproduce

Here's just a sample gcode of a nameplate

[LayerChangeTest.zip](https://github.com/user-attachments/files/19040143/LayerChangeTest.zip)

### Expected behavior

In comparison, my MK4S handles the same design flawlessly. The MK4S raises the Z-axis slightly and homes to the back. While I understand that the CORE One needs to lower the bed for magnet insertion, it seems to lower it excessively. Some tuning is needed to ensure that when the bed returns, the Z-axis is at the same layer it left off at.

crzykidd, 2025-03-01:
I have experienced the same issue on pausing prints and restarting.

CZDanol, 2025-03-01:
Hello, guys, we have identified the issue. It will not get into 6.2.2, but I will suggest putting it into 6.2.3, if there will be one.

crzykidd, 2025-03-01:
will 6.2.3 be available on the Core One?  Or is that following just the 6.3 line?

pipipipi2002, 2025-03-02:
Hi!

I faced this problem too last week. Unfortunately it happened when i paused it while printing on the first layer. When i resumed, the nozzle digs into the bed and damaged it. Contacted support last week but no solutions provided.

The-Printing-Sailor, 2025-03-03:
I have experienced the same problem. On returning after a pause to add a nut in a print. Z rises to high and nozzle scrapes across the print. CoreOne 6.3.0 + 10073.

CZDanol, 2025-03-03:
@crzykidd 6.2.3 will not be available for Core ONE. For Core ONE, there should be a 6.3.1 release soon-ish

nenovidenovic, 2025-03-06:
I'm experiencing the opposite - after pause print, my print head is to high and doesn't adhere. I made [a post about it](https://www.reddit.com/r/prusa3d/comments/1j1a1av/problems_with_pause_print/), with pictures, on Reddit and was redirected here.

LordMelkor, 2025-03-06:
I’m having the same issue as the user above. The nozzle is too high after the resume, so the layer does not adhere.

It’s very consistently repeatable. Made a [post](https://forum.prusa3d.com/forum/postid/738148/) on Prusa forums about it.

Logicwax, 2025-03-08:
I have the same issue, but the nozzle resumes too low.   This causes the gantry to make clicking sounds while under stress and the print goes to shit.    Resuming basically ruins the entire print.

73hhun73r, 2025-03-11:
I am seeing the same issue. The extruder is digging into the print, and getting layer shifts after resuming from pause.

CZDanol, 2025-03-12:
https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/47eb31be3dbec677cbfbf07c470e7fd7014093b6/lib/Marlin/Marlin/src/module/motion.cpp#L1057
This function disregarding MBL. It is used in marlin_server::unpark_head_ZE

No workarounds come to my mind.

PinkPandaKatie, 2025-03-15:
I tried to cherry-pick commit 167420c23 that fixes this bug onto 6.3.0, but there was another commit in 6.3.0 that I had to revert first.

Commit 44357a22f fixed a scenario with G28 being interrupted by opening the door, but I disabled the door sensor anyway.

I created a branch to fix it if anyone wants to compile it. 

https://github.com/PinkPandaKatie/Prusa-Firmware-Buddy/tree/z-unpause-fix-6.3.0

@CZDanol What's the policy on providing custom BBF files? I know not everyone has a build environment set up.

CZDanol, 2025-03-16:
@PinkPandaKatie We do not endorse it, but if anyone is willing to install it at his own risk, we're not blocking it.

The customers who would want to install the custom firmware will have to break their appendix on the board.

**DISCLAIMER:** Please be aware that we are not verifying custom builds and the authors can possibly sneak in some malicious code there, that could result in your printer getting damaged (if the damage was done as a consequence of installing an unofficial firmware, that is not covered under warranty) or your data being sent to the internet.

paulgregg, 2025-03-17:
Just adding that I too am experiencing this issue.   Any time I add a pause in via slicer (2.9.0 and 2.9.1) when I resume the print it seems the printer resumes at the wrong layer - I get blobs forming over where the magnets were inserted (perhaps the head is too high?) which very soon results in the head knocking the object off the print bed when it crashes into the hardened blobs.

e-basteln, 2025-03-21:
Rather than build a custom branch off of 6.3.0, I would like to see an official 6.3.x release that fixes this and various other problems in the Core One firmware. Any update when this might be forthcoming? The printer has been shipping for a while now, with what feels like more than its fair share of early issues. (VFAs and acoustic resonances, resume problem, crash upon opening the interlocked door...)

classenfeind, 2025-03-26:
Had the same issue ( with "pause at layer" and manual stop + resume in the menu), as a workaround i now use filamemt change comand instead of "pause at layer", this seems to work.

beelercoding, 2025-03-26:
I can agree with that. The "Add Color Change" command in the PrusaSlicer 2.9.1 worked also for me as a workaround fine.

> Had the same issue ( with "pause at layer" and manual stop + resume in the menu), as a workaround i now use filamemt change comand instead of "pause at layer", this seems to work.

pwarren42, 2025-03-29:
Same issue. Sometimes, pause and resume work OK. Other times it starts too low. Overall, it seems very difficult or impossible to remove print artifacts during the prints making them impossible to rescue when issues like filament globs appear.

Taubin, 2025-03-31:
> Customers who bought the Core One already went out on a limb to preorder a new product supporting Prusa, and are only getting endless frustration in return. You all need to get your priorities straight.

Honestly if you are that upset by it, return it and wait for a product you are happy with. Telling devs they "need to get their priorities straight" isn't going to get the fix to you any faster.

pwarren42, 2025-04-04:
I'm not sure how this works. Altaic's comment makes me wonder, is there a new firmware to download to fix this issue?

cryptomilk, 2025-04-04:
For those who don't find the release page: https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.3.1

Prusa-Support, 2025-04-05:
Exactly. Thanks for your patience.
FW 6.3.1 was published 3 days ago and is already posted on the [reference page](https://help.prusa3d.com/downloads) for the final releases.
It includes **[Fixed: Incorrect Z position when unpausing](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.3.1#:~:text=Fixed%3A%20Incorrect%20Z%20position%20when%20unpausing)** notes in the changelog.

Please allow me to close this issue as solved in firmware version  6.3.1 (CORE One) and 6.2.3 (older models).

Michele Moramarco
Prusa Research

closed 2025-04-05 by Prusa-Support

rodrigo2b, 2025-05-19:
I'm currently on FW 6.3.2 (just received my assembled core one in the mail), and I am encountering incorrect Z values after unpausing.