prusa3d/Prusa-Firmware-Buddy #4507: [ENHANCEMENT] Cancel Object - number items in view so you can discern when printing multiple of the same name

jurassic73, 2025-03-02:
### Printer model

MK4s

### Describe the enhancement

In the cancel object option on the MK4s, I can't tell which to select to cancel if I am printing multiples of the same item.

### Expected functionality

A view of the bed with objects on it would come up when clicking cancel object with one of the objects highlighted. If you turn the dial the highlight would move to the next item and so on until you've highlighted the item you want to cancel. Then you'd press the dial to cancel your item.

Tupson444, 2025-05-26:
This does seem difficult to implement, but it would be very useful. It should stay open at least to try to find an alternative, easier way to achieve a similar effect.

Perhaps the focus should be on PrusaSlicer, to make it always arrange objects from left to right and back to front, so when you scroll through objects to cancel, you would be able to know which object is which (when they don't have characteristic names). Currently I have to arrange objects manually to be able to cancel e.g. object 5, knowing it is the fifth object from the left, instead of waiting for printer to start printing it to "cancel current".

Tupson444, 2025-08-02:
Still wanted (some better way of selecting objects to cancel)