prusa3d/Prusa-Firmware-Buddy #4509: Change sheet profile until print starts

Saphir3D, 2025-03-04:
### Printer model

MK 3.5 with MMU3

### Describe the feature

I often change the steel sheets and I often forget to change the profile before I start a print. Several minutes can pass between starting the print in the menu and the printer starting to print after calibration, during which I would like to be able to change the profile of the steel sheet or even during the first layer.

### Expected functionality

Add the settings for changing the steel sheet profile to the print menu.

murk-sy, 2025-03-05:
I don't think changing sheets after calibration should be encouraged. But if the change is made before auto bed leveling starts, then is there really that much of a time loss just cancelling the print, changing the sheet in settings and restarting it?

Maybe having an option for a confirmation menu before printing (and possibly integration into Prusaslicer) would make sense, and that would also apply to newer models with nozzle probing.

Similar issue to that, except with a setting to ask before every print rather than only after filament change:
- https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3465

Saphir3D, 2025-03-05:
If I'm right, the print sheet profile is not much more than a preset value for Live adjust Z. So technically this should be possible both during calibration and during printing. It doesn't make sense from the second layer onwards, but that's another story.

closed 2025-05-12 by github-actions[bot]