prusa3d/Prusa-Firmware-Buddy #4511: [ENHANCEMENT] Allow the lights to turn off fully when CORE One not in use

Taubin, 2025-03-04:
### Printer model

CORE One

### Describe the enhancement

When the printer is idle for an  hour or so, it would be nice for the lights to turn off fully. I use the printer in a room next to where I sleep, and the light being on at all times is very distracting and makes it difficult to sleep. 

It would be nice to be able to select an idle time for them to turn off completely until the printer is "woken up" by either the screen or Prusa Connect. A few seconds for the printer to come back online (or the light to come back on) is just fine. 

### Expected functionality

Set an option in the settings to turn off the lights when the printer has been idle for 15, 30 and 60 minutes.

Having this setting enabled would turn off the lights on the printer until a print job is sent, the screen is touched, or the door is opened. Similar to how the lights dim and undim now, but instead would turn off entirely.

legomind, 2025-03-05:
Similar to #4146

CZDanol, 2025-03-07:
Hello, thanks for the idea. We will consider it. Internal ticket: BFW-6880

CZDanol, 2025-03-07:
Btw @Taubin XL and C1 do share the code handling the chamber dimming.

Taubin, 2025-03-07:
Thank you very much @CZDanol 

It's also good to know they share code for the chamber dimming. You learn something new every day. Have a great day.

Logicwax, 2025-03-08:
or how about just a button on the main interface to toggle chamber lights?   having to go deep into the settings menu to adjust them to 0% seems kinda ridiculous.

bkerler, 2025-03-23:
There is a PR #4174  which is also compatible for Core One, which also adds auto display dimming.

jim-p, 2025-04-23:
Having them dim/turn off while idle is good, but I would also like an option to keep the lights at full brightness while actively printing (or even when any heaters are running).

Some cameras like the ESP32CAM w/wide angle lens don't deal very well with the lower light level which makes monitoring the printer more difficult than it should be.

CZDanol, 2025-04-24:
You can already turn the dimming off.

jim-p, 2025-04-24:
> You can already turn the dimming off.

No idea how I missed that under the UI menu. Lights sort of blur the line between Hardware/UI and all most of other UI options are about the screen. Thanks!

henxdl, 2025-07-08:
Also, what's the point of remotely uploading files if I have to go into another room to turn it on, come back to my PC and start the print?

PyroDevil, 2025-07-18:
I tested it and I like it.

However can the RBG status bar be included in the dimming as well? Currently we can only enable or disable it. Maybe have three settings: On, Off, and Dimmed?

tjhawkey, 2025-08-08:
I opened a bug issue relating to the dimming function in 6.3.4 and 6.3.5 alpha NOT working as intended. It got closed with the comment that it's a duplicate to this issue. I think that's a bug, not an enhancement.

t-makaro, 2025-08-29:
I want to be able to turn off the chamber light manually mid print with a quick toggle rather than going through like 3 menus and then scrolling the brightness.

azakhary, 2025-11-06:
Ye i wana be able to have it ON consntaly, so i can print when not home. but then if i leave it on it is fully ON like led's and who knows even maybe fan (although not sure). anyway, "sleep mode" would be nice

onionhammer, 2026-02-18:
@Prusa-Support this should be reopened IMO until the touchscreen is dimmed/turned off also when idle, leaving the touch digitizer on

NateroniPizza, 2026-02-19:
> [@Prusa-Support](https://github.com/Prusa-Support) this should be reopened IMO until the touchscreen is dimmed/turned off also when idle, leaving the touch digitizer on

The issue they referenced (#660 ) is specific to that - this would be a duplicate of re-opened.