prusa3d/Prusa-Firmware-Buddy #4512: [BUG] Search bar only links to first result.

p41nte4, 2025-03-05:
### Printer model

n/a

### Firmware version

n/a

### Upgrades and modifications

n/a

### Printing from...

n/a

### Describe the bug

When searching for anything in the search bar with more than 2 results the first result is the only result it takes me to when double clicking the first time. usually the second click on a different result works but coosing another result has the same effect. this has been happening since at least the last stable build. search bar will display the correct results but will not go to them, except for the first one the first time.

### How to reproduce

search for anything, and ensure more than 2 results. double click 2nd 3rd etc. result and takes you to first result only. single clicks ae different sometimes takes 2 or 3 attempts but does eventually work. 2nd attempt on same result usually works but click another result and it stays on the current result.

### Expected behavior

clicking a result should take you to the correct result first time. this may be linked to the click bug where text cannot be input on first attempt.

### Files

do not have facility for video, screenshot wont show it.

CZDanol, 2025-03-05:
What search bar are you talking about?

p41nte4, 2025-03-05:
doh im sorry. please delete thought i was still  in prusaslicer git