prusa3d/Prusa-Firmware-Buddy #4515: [BUG] "Please wait" screen does not fully hide footer icons on Mini

murk-sy, 2025-03-07:
### Printer model

Mini+

### Firmware version

6.2.2

### Upgrades and modifications

Filament sensor

### Printing from...

n/a

### Describe the bug

When running "Auto Home" Or "Mesh bed leveling" in the Control menu (or others that use the same overlay), the very edges of the footer are still visible

[image]

Simulated:

[image]

### Expected behavior

a) Cover up the remaining pixels, or
b) Show footer during this operation (is there a particular reason why it's not visible? It's normally visible when the operation is triggered during a print)

CZDanol, 2025-03-10:
Internal ticket: BFW-6894

murk-sy, 2025-05-15:
Still active