prusa3d/Prusa-Firmware-Buddy #4516: [BUG] Overcurrent in MMU error when resetting while filament loaded in Nextruder

guiweber, 2025-03-07:
### Printer model

MK4S

### Firmware version

6.2.2+8853

### Upgrades and modifications

MMU3

### Printing from...

Prusa Connect

### Describe the bug

The MMU3 reports an error when powering up the printer if filament was loaded in the Nextruder when it was reset or shut down.

It seems that the loaded filament physically blocks the initialization routine of the MMU, creating an over-current situation.

This might be specific to the printer state when the reset was performed or might be more general, I didn't do further tests than the repro steps provided below.

### How to reproduce

1. Start a print with the MMU3
2. Stop the print before the end, the filament will stay loaded in the Nextruder
3. Start a new print, the printer will offer to unload the filament first
4. Reset the printer at that point, without unloading the filament
5. The over current protection will be triggered upon restart

### Expected behavior

Upon power up, any filament loaded in the Nextruder should be detected and unloaded if necessary before initializing the MMU.

CZDanol, 2025-03-10:
Hello, we have recentlay released 6.2.3-alpha which should address some MMU overcurrent issues: https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.2.3-alpha1

Would you be willing to verify that the issue still persists in this version?

guiweber, 2025-03-25:
@CZDanol sorry I didn't have time to check before now. I did tests on 6.2.3 release, resetting from multiple parts of the UI with a filament loaded and couldn't reproduce the issue. Seems fixed!