prusa3d/Prusa-Firmware-Buddy #4517: [BUG] Printing stuck, the rebooted

Pitel, 2025-03-07:
### Printer model

Mini

### Firmware version

6.2.2

### Upgrades and modifications

Filament runout sensor

### Printing from...

Prusa Connect

### Describe the bug

I was printing, and filament ran out. So I put in in a new one from scraps. It printed for like 5 seconds then got stuck (like, no movement, just fans running). So I tried pausing and resuming, and it resumed correctly. But then the filament ran out again, and same story, except now it rebooted after resuming

### How to reproduce

Since it got stuck filament ran out 2 times, I think that's how you can reproduce it.

### Expected behavior

Continue printing normaly

### Files

Dump.bin sent over email.

CZDanol, 2025-03-10:
Internal ticket: BFW-6893

Prusa-Support, 2025-03-11:
Thanks for providing the dump file.

Please don't forget the reference number, but don't worry, I was able to locate the file already.
The reference code is provided as automatic reply in the acknowledgment email.

PR31502689282

Michele Moramarco
Prusa Research

closed 2025-05-19 by github-actions[bot]