prusa3d/Prusa-Firmware-Buddy #4520: [BUG] Unreliable start up sequence due to chamber temperature or nozzle cleaning error

huskyte, 2025-03-09:
### Printer model

Core One

### Firmware version

6.3.0

### Upgrades and modifications

None, factory build

### Printing from...

Prusa Connect

### Describe the bug

I own two C1 (assembled) and both of them show the same issue over and over again. Either chamber temp heating error or nozzle cleaning error (solely printing with ASA/PP/PC/ABS). My best bet is the tolerances are to tight. 
The chamber heating works if restarted or preheated. But coming from 18 °C I always run into the problem. And as it is not actively heated/hence not really controlled this is insane. Just let the firmware check for the current temp and let it wait for two or three minutes. Or heat it up to 30 °C and then let the procedure take over.
As for the nozzle cleaning: Probably more complex: I noticed the problem is far more prominent if a print was skipped. Probably not enough/no retraction before starting a print. But it happens regularly. I need to clean the nozzle of nearly no material and get rid of the 20 something spots it marked on the print bed. What is also noteworthy: The tapping happens at a part of the bed where no heat bed is beneath. Sems a bit silly. That part probably bends. Also, why do I need to physically go to the printer if it is just a stupid retry without solution. Implement a function to repeat that 5 cm to the left in connect. Hence no cleaning of the bed required. And after the first cleaning fails it is obvious that 1 mm retraction might as well help. So, needs to be in the code. And soften the tolerances. Same as with analytical balances: there are settings for different environments. If I have another Core One next to it hammering and vibrating, well, this procedure becomes close to useless within the limits it is trying to achieve. Let us choose how much precision we want to have.

### Files

I own two C1 (assembled) and both of them show the same issue over and over again. Either chamber temp heating error or nozzle cleaning error (solely printing with ASA/PP/PC/ABS). My best bet is the tolerances are to tight. 
The chamber heating works if restarted or preheated. But coming from 18 °C I always run into the problem. And as it is not actively heated/hence not really controlled this is insane. Just let the firmware check for the current temp and let it wait for two or three minutes. Or heat it up to 30 °C and then let the procedure take over.
As for the nozzle cleaning: Probably more complex: I noticed the problem is far more prominent if a print was skipped. Probably not enough/no retraction before starting a print. But it happens regularly. I need to clean the nozzle of nearly no material and get rid of the 20 something spots it marked on the print bed. What is also noteworthy: The tapping happens at a part of the bed where no heat bed is beneath. Sems a bit silly. That part probably bends. Also, why do I need to physically go to the printer if it is just a stupid retry without solution. Implement a function to repeat that 5 cm to the left in connect. Hence no cleaning of the bed required. And after the first cleaning fails it is obvious that 1 mm retraction might as well help. So, needs to be in the code. And soften the tolerances. Same as with analytical balances: there are settings for different environments. If I have another Core One next to it hammering and vibrating, well, this procedure becomes close to useless within the limits it is trying to achieve. Let us choose how much precision we want to have.

CZDanol, 2025-03-10:
Hello, you have mixed several suggestions, bugreports and complaints into a single issue, which is making it impossible to address individual things separately. Please dedicate some time into organizing your thoughts and then possibly open separate issues for separate things.

Also:

1. What are you referring to by "chamber temp heating error"? If you are referring to "failing to reach chamber temperature":
  1. We will be increasing the limit in the next version
  2. The error disappears itself if the printer manages to reach the temperature after all

closed 2025-03-10 by CZDanol

huskyte, 2025-03-10:
Separated for nozzle cleaning error in #4523 . Thanks for the hint.
As for chamber heating: Well, I won't file a report here, but if it is the case (warning, disappears automatically) Prusa Connect needs to differentiate between alarms and warnings. A pop up in Connect makes me walk to the printer and check/do something. A warning would be okayish (stating something like a slower heat up, maybe in another color). Although I think the firmware should be capable of handling anyting between 10-40 °C as an expected result.