prusa3d/Prusa-Firmware-Buddy #4521: [ENHANCEMENT] Option to set higher temperature during filament change

timur-tabi, 2025-03-10:
### Printer model

MK3.5s

### Describe the enhancement

When the filament runs out, the printer prompts for a filament change.  In some cases, it appears that the old filament breaks before it is fully ejected.  The break occurs just below the geared shaft.  This makes it almost impossible to insert the new filament.

Part of the problem is that I tend to print with temperatures on the low side.  E.g. I print PLA at 195C.

I have found that an easy solution to this problem is to raise the temperature significantly (e.g. 290C when changing PLA), which allows me to ram the new filament into the old.  If I do not raise the temperature, the old filament will not budge.  But at 290C, the heat transfer through the filament more, allowing more of it to melt and unjam.

So what I would like is a firmware setting that lets me specify a "filament change temperature", either as a set value, or per-filament, or something.  It could be per-filament M command that I send as a custom startup.

murk-sy, 2025-03-10:
in `Filament > Manage Filaments` you can create a custom PLA filament with higher print temperatures which are then used for filament change, but not actual printing (since that's slicer controlled). You would just need to avoid using that profile for first layer calibration and I believe that would essentially resolve your request.

timur-tabi, 2025-03-10:
First, there is no "Manage Filaments" menu on my MK3.5s.

Second, I don't think this is going to work because the default temp for PLA is 215, but when I tried to change the filament, it stayed at 195 during the entire operation.  That's why I filed this request.

murk-sy, 2025-03-10:
I wasn't sure if there was some reason this features was missing from 3.5 but that's not the case. You're just running old firmware.
https://help.prusa3d.com/article/lcd-menu-mk3-5-s_815309

Update to at least 6.2.0, which added this functionality.
Latest stable is 6.2.2: https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.2.2

ttabi, 2025-05-14:
I'm still interested in this feature.

closed 2025-07-22 by github-actions[bot]