prusa3d/Prusa-Firmware-Buddy #4523: [BUG] Unreliable start up sequence due nozzle cleaning error

huskyte, 2025-03-10:
### Printer model

Core One

### Firmware version

6.3.0

### Upgrades and modifications

factory default

### Printing from...

Prusa Connect

### Describe the bug

I own two C1 (assembled) and both of them show the same issue over and over again. 

This is a the successor of issue #4520 

Nozzle cleaning fails at the start print sequence: I noticed the problem is far more prominent if a print was skipped. Probably not enough/no retraction before starting a print. But it happens regularly. I need to clean the nozzle of nearly no material manually and get rid of the 20 something spots it marked on the print bed. And let it run again, which fails again with high probability. 

What is also noteworthy: The tapping happens at a part of the bed where no heat bed is beneath. Sems a bit suboptimal as this part is probably rather unstable. Also, cleaning the nozzle is not the problem (as shown on the display), but rather the oozing.

### Expected behavior

Implement a function to repeat that 5 cm to the left, initiated automatically or by Prusa connect. Hence no cleaning of the bed required. And after the first cleaning fails it is obvious that 1 mm retraction might as well help.

Soften the tolerances. Same as with analytical balances: there are settings for different environments. If I have another Core One next to it hammering and vibrating, well, this procedure becomes close to useless within the limits it is trying to achieve. Implement 2+ modes for z probing precision, choosable via menu.

Alternatively rethink the start up sequence. There might be alternative ways to come to a proper z calibration that I am not aware of.

loudurdee, 2025-03-18:
Same issue here. Repeatedly failing at the Nozzle Cleaning stage of starting a print. At first it was only when I tried printing PETG, but now has done it on PLA as well. 

I can usually get past it by pausing the print and restarting it, but sometimes takes multiple cycles of pause and resume before moving on to probing the bed. 

I may not be passing the nozzle check by doing this, but just skipping past it by interrupting the sequence. That's how it seems anyway, as it doesn't keep trying to clean after resuming, it just bumps the bed once and moves on to probing. 

After getting past the "Nozzle Cleaning" all my prints have come out phenomenally. Layer for layer looks great and no stringing, etc. It's only this starting nozzle check that fails.

gajowy, 2025-03-18:
Same problem here, nozzle cleaning procedure fails multiple times usually 3x, sometimes 4x. Happens with all materials, especially with TPU, PLA least, happens on satin finish bed and smooth as well. 
No signs of material on the nozzle but the printer tries to clean it anyway.

BarryWA, 2025-03-25:
Nozzle cleaning error fail with Core One, firmware 6.3.0.  When printing Prusament PC-CF multiple cleaning fails.  Purge is fine, nozzle is clean.  Very difficult to get past.  Suggest "continue" or "ignore" .
After multiple attempts, I can get past, print quality OK.

BarryWA, 2025-03-31:
I can confirm that with the same nozzle and print bed PLA passed the nozzle clean without problem.  It seems that filament stiffness and or temperature play a part in the issue.

BarryWA, 2025-04-08:
Happy to report that I had an easy nozzle clean with Prusament PC-CF using the 6.3.1 update.    Did a long purge ahead to be sure nozzle was clean and extruding well.  Two or three bumps and started the print.  ObX .04 nozzle, temp 285C, Prusa Slicer standard.  Filament dried and printing out of dry box.  Just one attempt, but seems good now.  

Update; after printing PC-CF and changing to PLA I had a nozzle clean fail.  I did a long purge which solved the issue; perhaps residual filament.  Early days on the firmware update, but seems to have solved the issue.  I still would like the option to "continue" and print after nozzle clean "fail" that seems to not be a nozzle issue but rather calibration issue.

CZDanol, 2025-04-10:
There will be an option to ignore the nozzle cleaining failure in 6.4. It is already written, but too much of a code to backport.

PedorthicArt, 2025-04-10:
Seems my 'nozzle cleaning error' disappeared once I upgraded the firmware to 6.3.1
Also changing the nozzle from HF to standard brass seems to be the solution to succeed to print flexible TPU (stiffness < 95A) on Core One

BarryWA, 2025-04-10:
> CZDanol
>  left a comment
> (prusa3d/Prusa-Firmware-Buddy#4523)
> There will be an option to ignore the nozzle cleaining fail in 6.4. It is already written, but too much of a code to backport.

Thank you very much for letting us know.  Thanks for your efforts to smooth out the printing process.   I suspect that will be a big help, although 6.3.1 has helped.  

Barry

zeus86, 2025-04-20:
Same nozzle cleaning error on factory-built C1 with 6.3.2. This is annoying AF, because it persistently fails even when the nozzle is perfectly clean. just before the nozzle cleaning, the printer ramps the temp a little, so that it always oozes, and opening the door within the cleaning step causes the printer to crash 9 out of 10 times. this recently hit me on freshly dried prusament galaxy black PETG.  please make this (in the current state anyways useless test) at least skippable!

CZDanol, 2025-04-22:
Could you please send us a log from the nozzle cleaning procedure?

Turn on logging to a file in Settings → System
Run the nozzle cleaning (possibly as part of the print)
When it fails, abort the print
Turn off the logging
Share the log.txt file

Thanks :)

PrintFandango, 2025-04-22:
This is very frustrating as it normally takes several attempts to get the printer to start printing. Filament is dry and there is no oozing for me. If there is oozing there is 100% guarantee either nozzxzle cleaning or bed proving will fail, but even with no oozing, it sometimes fail. I can simply 'retry' and the it could move forward next time... sometimes it could take 3-4 attempts.

Just now, did nozzle clean - prusa connect said printer needs attention - but the printer went on to do bed probing, then continued bed probing and finally failed bed probing and paused. Very extrange behaviour and very frustrating.

Log attached

[log.txt](https://github.com/user-attachments/files/19850557/log.txt)

CZDanol, 2025-04-23:
Thank you @PrintFandango for the log. I'm sorry about the issues you're having. The log didn't confirm my suspicion on what could be causing it on the firmware side. It rather seems to indicate that there is something in the hardware. I'd probably recommend you to check:

1. That everything on the print head is properly tightened up. This could possible be caused by some components flexing when they shouldn't.
2. Same with print bed.
3. Verify that the printer passes the loadcell selftest - there might be some noise on the loadcell.
4. Just out of curiosity, you can try bringing the printer to another room and plug it into another wall outlet.

PrintFandango, 2025-04-29:
none of these tips helped. the issue only exists when anything other than PLA is used.

Prusa-Support, 2025-05-16:
I seem to understand that some of you may often be changing materials in a more or less wide range of material temperatures, and surely printing in variable ambient temperatures too.
Besides potential hardware concerns - https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4523#issuecomment-2823316455 - I wonder if the following helps with reducing the chances of nozzle cleaning problems. 

- Perform a cold pull before switching to a very different material temperature.
  - Please do in either case, from lower to higher temperature, or vice versa!
- As soon as new filament is loaded, let the nozzle ooze for several seconds before removing purged material.
  - Several seconds and not several minutes! Plastics may harden and degrade in case of long heat exposure.
- Dry the filament overnight at 40°C, ideally with an active fan.
  - This is a generally safe temperature. Consult the filament manufacturer before drying at any higher temperatures.
- Let the printer preheat at target temperatures several minutes before the start of the print.
  - Especially useful in cool environments, but possibly observe the operating temperature stated in the manual. 

Please take your time to test across several print scenarios and let us know if any of these tips were useful to you.

Michele Moramarco
Prusa Research

zeus86, 2025-05-16:
A cold pull seems to be totally unneccessary and impractical. to me this happens primarily when changing from PLA to PETG and then Back to PLA. However, PLA and PETG are not "very different", at least not from their parameters to print with, and second, for a cold pull you have to remove the top acrylic sheet of the core-one, which is impractical by itself, but VERY impractical especially on the core one, because of those single-use-plasic-rivets, and the fact, that you have to disassemble everything above the extruder (the tubing and such). Doing this on every material change is bonkers.

rfilgas, 2025-05-21:
Any word on when 6.4 is due to release?

jonathanalevi, 2025-05-28:
Joining in here. I have this issue on both of my Core ONEs. If there is ANY filament anywhere near the nozzle, it will cause a nozzle cleaning issue. It seems the current sequence just isn't sufficient to clean anything but the tiniest imperceptible amount of filament on the nozzle. I'd say we need a proper silicone brush or something, but, maybe it can be fixed with a dragging sequence... either way, the way it's set up now doesn't work.

BarryWA, 2025-05-29:
I'm running the current firmware and presently experiencing no problems with nozzle cleaning.  I'm using a hardened OX .04 nozzle with a variety of filaments and print beds.  Filaments: PLA from Prusa and Hatchbox, Prusa PC CF and PP CF.  Print beds, PEI smooth, textured Prusa, Prusa PP, and Garolite.  I did have multiple nozzle fails early on, the firmware updates seemed to have helped.  Only a rare fail, which clears on second try or with pause, retry.

nschllr, 2025-06-20:
I’ve had this issue for about 1.5 months. Most of the time, it took two attempts to get past the nozzle-cleaning stage for every print (which is super annoying) — and I was always printing with PETG. However, yesterday I swapped my nozzle from a high-flow one to a standard one, and for the first print after the change, the problem didn’t occur. Now, after that print, the printer is no longer able to get past the nozzle-cleaning stage at all.

Firmware: 6.3.3

mkyral, 2025-07-14:
Yesterday I've spent almost one hour till print finally starts. With combination of Homing issue (https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4582) it is nightmare.

I've printed from PETG. When cleaning failed, I've noticed to late and nozzle was already cold. There is no beep?

So I hit retry, but printer starts a new homing sequence (why?) and it took forever. I've reset printer - homing ok, nozzle cleaning failed again. I've used brass brush, cleaned as much as possible, hit repeat, but again - homing took forever. Reset again, new cleaning again fails…

Somehow I was finally able to pass this and print starts, but it is too far from simple hit of "Print" button.

3d-gussner, 2025-08-08:
@anderssonsdev If I understand it correctly you have issues with gcode generated by Printables online slicer. If so, please contact PrusaSupport and provide the two gcode files, one generated by Printables and the other using PrusaSlicer. I would like to see what difference is. And if there is something that may cause that issue. Please also ask them to notify me.

You can also Use PrusaSlicer to convert bgcode to gcode if you want to compare these.

JeffAGreen, 2025-08-11:
I had the nozzle clean failure after I built the kit. Spent a lot of time with tech support. It's apparently factory and kit. They think it's caused by damage/kinks in the Nextruder or load cell cables. They sent me out a new load cell and new Netruder cable and after installing, the issue is gone. I did find a crimp in the Nextruder cable after I had it all apart. They seem to be pretty aware of this issue. One thing I did, in the meantime, to trick the prints into working was to drop the hotend slightly. Not sure why it worked, but it did. No more nozzle fails and it printed awesome. But I still did the fix, because it's not ideal to have the hotend out of position.

wizzlebippi, 2025-08-14:
Some evidence of a firmware bug regarding nozzle cleaning.  I converted my MK4S to a Core One, and updated from 6.2.4 to 6.3.4.  No settings changed, only calibrations ran.  I had the nozzle cleaning bug initially, but it only appeared once.  After about 10 prints, the nozzle cleaning bug struck hard and my printer was inoperative.  I ran through all the hardware checks in the FAQ, work with support for a while, but nothing changed.  I then decided to perform a full reset on my printer, reinstalling the same 6.3.4 firmware file.  It ran through the initial calibrations, and I was able to print again with no drama.

3d-gussner, 2025-08-14:
@wizzlebippi Thanks for the information.
I would always suggest a factory reset after big change like the conversion from MK4S to CoreOne.
On the MK3/S users often had issues after modding their printer (REVO) and some previous MK3S/+ eeprom values caused issues with the REVO build.

Prusa-Support, 2025-09-16:
Once again, I would like to encourage you to have the printer and print conditions double-checked through the "simplified" [checklist above](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4523#issuecomment-3217155864), and with the help of our Customer Support - chat or email, we are totally available.

However, I would like you to know that a new firmware pre-release [6.4.0-alpha](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.4.0-alpha) is now available, offering the possibility to recover from nozzle cleaning, or to skip the process and continue after a quick nozzle check.

Michele Moramarco
Prusa Research

graelo, 2025-09-16:
Thanks Michele, I went over the list above as soon as you posted it, and for me, retightening the heater block and ensuring no two wires were stranded against another made the nozzle cleaning errors disappear: not a single one since then (I was having them before each print). I recognize there might be different situations though. Thanks!