prusa3d/Prusa-Firmware-Buddy #4524: [BUG] Starting a print sometimes generates "Attention" status and iOS notification when attention is not needed

neuron-whisperer, 2025-03-11:
### Printer model

MK4S

### Firmware version

6.2.1

### Upgrades and modifications

None

### Printing from...

PrusaLink

### Describe the bug

When a print is started via PrusaLink, the printer sometimes shows an attention of "Status" for about 15 seconds. Frequently, this alert also causes the iOS Prusa app to generates an iOS notification:

[image]

However, no attention is required at this time - the printer is just getting started - the print routinely continues without intervention. Although this bug is not a functional problem, it is still an annoyance, since the printer sometimes does require attention at the start of a print (e.g., stuck filament or runout issues).

Speculation: This behavior might be due to the printer displaying a notification of a new firmware update. I'm aware that a new firmware was released a few days ago, and as I haven't had time to update it yet, I manually clear the notification when I am directly interacting with the printer. But PrusaLink does not have any facility to notify the user of a new firmware update (nor, I believe, does Prusa Connect), so the UI event is apparently lumped into the "Attention" category. However, I am not certain of this conclusion - I believe that the "Attention" error was occurring before the firmware notifications were first presented - so that might not be the cause.

### How to reproduce

Start print. Observe notification.

### Expected behavior

The printer should generate "Attention" notifications only when user intervention is required.

neuron-whisperer, 2025-03-17:
An update: I flashed both printers with the new firmware, but the "Attention" alert is still happening. I confirmed that the printers are no longer displaying the "New Firmware Available" message, so I have no idea why this is occurring.

Graloth, 2025-04-01:
Can confirm that this is also happening for almost every print on my XL.

neuron-whisperer, 2025-05-01:
Note - this is still happening on 6.2.4. I just flashed the printer, and it installed correctly and rebooted. And when I start a print, I immediately got an "Attention" notification through the Prusa iOS app.

closed 2025-07-09 by github-actions[bot]

3d-gussner, 2025-07-09:
The application and PrusaConnect team is aware of some users reporting this.
As this isn't a firmware issue, at least it doesn't look like it is, we leave this github issue closed.

The best way to address issues with the PrusaApp and PrusaConnect is via chat support.

neuron-whisperer, 2025-12-06:
I suggest that this issue should be reopened.

The "Attention" prompt is now also being shown in PrusaLink at the start of a print, even when the print is progressing just fine:

[image]

This behavior in PrusaLink appears to be new as of the 6.4.0 firmware - I just upgraded before this print, and now it's showing this message. The display of the printer shows no problems and no corresponding "Attention" prompt, and the print itself is progressing just fine.

Noting this line of the 6.4.0 firmware readme:

> Changes and improvements

> Improved print status messages

...I presume that some part of the firmware generates the "Attention" notifications for both iOS and PrusaLink, and is incorrectly parsing some of the new status messages as being problems that require attention.

In any case, please reopen this issue so that the Prusa team can give it due attention, unlike the first time that I submitted it. Thanks.

Prusa-Support, 2025-12-19:
Please consider reporting via [the feedback button](https://prusaconnect.sleekplan.app/changelog?type=t66ba28b9947e7) available in Prusa Connect (with `mobile app` tag), or writing to app@prusa3d.com.

Michele Moramarco
Prusa Research

jim0020, 2026-02-24:
This happens to me every time I print from local storage. 

Core One + upgrade from MK4S. FW: 6.4.0+11974.

samuelT2, 2026-10-21:
Very annoying. Happens for me for EVERY print. Reported via feedback form in Prusa-App as well.

Core One + (Gen 2), FW 7.0.0, Prusa-iOS App 1.14.23+362