prusa3d/Prusa-Firmware-Buddy #4525: [BUG] Stuck filament detection keeps triggering while printing pre-sliced benchy with brand-new pre-assambled MK4S +MMU3 with brand new Prusament PLA

5jvm0u4, 2025-03-11:
### Printer model

MK4S +MMU3

### Firmware version

6.2.2 & 3.0.3

### Upgrades and modifications

Brand-new, Stock, Pre-Assambled

### Printing from...

Provided USB dirve with provided presliced gcode

### Describe the bug

Stuck filament detection kept triggering while printing pre-sliced benchyrule-benchy with brand-new pre-assambled MK4S +MMU3 with provided brand new Prusament PLA.
This is because presliced files are sliced with CHT nozzle limitations, but when buying a pre-assambled MK4S with MMU3, normal nozzle is pre-installed, thus not providing enough melting throughput.
People buying pre-assambled printers should not be assumed to know this and they would assume the printer is broken, like my colleague who ran into this problem, it's hurting Prusa's reputation of being premium.

### How to reproduce

Print the given presliced benchyrule benchy file with a normal nozzle instead of a CHT

### Expected behavior

Ship the USB stick with presliced files that matches MMU default nozzle, at this price point, in my humble opinion, this should be expected.

CZDanol, 2025-03-11:
Hello, I will pass this to the relevant department.

When trying to print a HF gcode with a non-HF nozzle, you should get a warning at the beginning of the print. Did you get it? Maybe we forgot to re-slice the gcodes with a newer version of slicer that emits this information.

5jvm0u4, 2025-03-11:
I also expect the warning, that's why I couldn't figure it out for the first two attempt.

AndreasAusD, 2025-03-12:
Hi, 
I had a similar Problem with a self assembled MK4S.
I think the Benchy gcode is created for the high flow nozzle that is shipped in the MK4S, but with the MMU3 the normal nozzle is recommended.
Maybe you want to check the assembly manual the check what nozzle you are using

https://help.prusa3d.com/de/guide/9a-mk4-s-mk3-9-s-nextruder-mod_683106#765184

5jvm0u4, 2025-03-12:
Yes it is.

5jvm0u4, 2025-05-12:
This bot doesn't really make any sense does it? If it's still not solved, and the issue remains, why would there be any updates from my side?

closed 2025-07-22 by github-actions[bot]