prusa3d/Prusa-Firmware-Buddy #4526: [FEATURE REQUEST] Maintenance reminders

albjerre, 2025-03-12:
### Printer model

XL, Mk4, CoreOne, Mini

### Describe the feature

I suggest that Prusa makes a simple maintenance reminder feature in either the firmware or in PrusaConnect where it notifies you as a user when it is time to do maintenance. 

This would greatly increase the likelihood that I as user will perform the maintenance in time before I run in to problems. It is especially useful if you use the printers in a setting like a company where there are many printers and user.

I recently also discovered that Prusa had updated the maintenance procedures for my Prusa XL. But I had not seen this and therefore didn't know that I should perform this maintenance. By having an automatic maintenance reminders I would always know what is expected to keep my printer running smoothly.

### Expected functionality

The intervals for the Prusa XL are already defined in this article:
[https://help.prusa3d.com/article/regular-printer-maintenance-xl_401237](https://help.prusa3d.com/article/regular-printer-maintenance-xl_401237)

So, Prusa just have to make simple triggers based on either days passed or print hours passed since the last maintenance was done. Then I as a user can check off when I have done the maintenance, and the counters would reset.

If the counters were visible in PrusaConnect or on the printer UI, I could also check once-in-a-while to see if maintenance is coming up soon.

At my work we have MarkForged printers where they have a maintenance schedule utility for this: [https://support.markforged.com/portal/s/article/Maintenance-Schedule-Utility](https://support.markforged.com/portal/s/article/Maintenance-Schedule-Utility)

Tupson444, 2025-05-12:
This is a good idea, still wanted

albjerre, 2025-05-12:
Hereby updated and elaborated

dgarcu, 2025-08-27:
I had the same idea. I don’t manage many printers, but since I only use the machine from time to time, I’m not actively tracking the printer’s working hours — and therefore not doing maintenance accordingly.

It would be great to have a simple reminder system, like “lubricate rails” or “check belt tension,” based on the actual hours the printer has been running.

It would even be helpful to have a maintenance log, similar to what’s used for vehicles — showing when specific components were last checked, whether any consumables were replaced, and so on.

3d-gussner, 2025-08-27:
> It would be great to have a simple reminder system, like “lubricate rails” or “check belt tension,” based on the actual hours the printer has been running.

To use minimal resources maybe two message would be possible for a small check ~200h and big check ~600h with one link to https://help.prusa3d.com/article/regular-printer-maintenance-core-one_829710 or https://help.prusa3d.com/article/regular-printer-maintenance-mk4-s-mk3-9-s_419000

>It would even be helpful to have a maintenance log, similar to what’s used for vehicles — showing when specific components were last checked, whether any consumables were replaced, and so on.

This will use too much resources and users would need to confirm each of the step. I don't think that is very practicable.
On the web site a checklist is easy to create that users can follow.

albjerre, 2026-05-18:
Still relevant

albjerre, 2026-08-16:
@3d-gussner Please reopen this.

albjerre, 2026-10-19:
Still relevant